Clon1998/mobileraker #324: Thumbnail picture in notification not oriented correctly

Kubajz0, 2024-01-22:
Hello,

I am having issue that my camera feed needs to be rotated 180 degrees in order to be displayed correctly. This works just fine for the video stream however if the camera takes a picture for the thumbnail notification it is not rotated as per setup. Please see attached images. I did not find any evidence/option how to set this up just for the snapshot alone. It does not matter if I choose Rotation = 180 deg. or options (Flip vertical + Flip horizontal), both options rotate the video feed but not the thumbnail notification picture. Can you please take a look and maybe add option to rotate for the snapshots for the notification thumbnails?

### Screenshots
[image]
[image]

### Version Information

- Device-OS (iOS): [17.2.1]
- Mobileraker: [v0.4.0-41 for RPi / 2.6.10 Mobile device]
- Klipper: [v0.12.0-75]
- Moonraker: [v0.8.0-267]

### Debug Logs

[mobileraker_2024-01-22T09-25-50.log](https://github.com/Clon1998/mobileraker/files/14008319/mobileraker_2024-01-22T09-25-50.log)
[mobileraker_2024-01-22T10-48-24.log](https://github.com/Clon1998/mobileraker/files/14008320/mobileraker_2024-01-22T10-48-24.log)
[mobileraker_2024-01-22T10-52-53.log](https://github.com/Clon1998/mobileraker/files/14008321/mobileraker_2024-01-22T10-52-53.log)
[mobileraker_2024-01-22T10-53-42.log](https://github.com/Clon1998/mobileraker/files/14008322/mobileraker_2024-01-22T10-53-42.log)
[mobileraker_2024-01-22T10-57-10.log](https://github.com/Clon1998/mobileraker/files/14008323/mobileraker_2024-01-22T10-57-10.log)

---

#### Checklist

- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-01-22:
Hey,
currently, the webcam settings (Flip, Rotations) are not synced to the companion. You will need to configure them yourself in the config file. You can find out more at the [config reference](https://github.com/Clon1998/mobileraker_companion/?tab=readme-ov-file#companion---config).

closed 2024-01-22 by Clon1998

Kubajz0, 2024-01-22:
> Hey, currently, the webcam settings (Flip, Rotations) are not synced to the companion. You will need to configure them yourself in the config file. You can find out more at the [config reference](https://github.com/Clon1998/mobileraker_companion/?tab=readme-ov-file#companion---config).

Thank you very much for this information I will look into that.