Clon1998/mobileraker #327: Google account OAUTH

GroteGehaktBal, 2024-01-25:
---

## **Feature Request**

### Problem Description
I'm currently facing challenges with integrating Cloudflare tunnel authentication for connecting via the MobileRaker app. After adding Google account authentication to the Cloudflare tunnel, the app is unable to complete the OAuth flow required for Google authentication, rendering it unable to connect on the remote connection.

### Proposed Solution
I propose the implementation of a feature that allows for an alternative authentication method when connecting through Cloudflare tunnels. This could be in the form of a specific API token or an authentication method that is secure and works with CloudFlare. This feature should ensure secure access while maintaining the convenience of app connectivity.

### Alternatives Considered
VPN or Direct Connection: Setting up a VPN or a direct connection for trusted devices, bypassing Cloudflare for these specific cases.

### Additional Context
Unfortunately, I don't have screenshots or examples to provide at this moment. However, the main issue arises from the OAuth flow required for Google authentication, which is not supported by the MobileRaker app. The solution needs to balance security with ease of access for trusted applications.

---

I appreciate your consideration of this feature request. Please correct me if this feature is already available...

Clon1998, 2024-01-25:
Hey,
Just for clarification. You are talking about the Google auth that is now required for your cloudflare tunnel and not anything related to the inner workings of mobileraker?

GroteGehaktBal, 2024-01-25:
> Hey,
> Just for clarification. You are talking about the Google auth that is now required for your cloudflare tunnel and not anything related to the inner workings of mobileraker?

Correct, I added cloudflare as an extra layer of protection but after enabling the authentication with google the mobileraker app does not connect anymore.

When I open the page on a browser it will first show me the login for the cloudflare tunnel google account and after logging it it will send me to the fluidd dashboard.

Clon1998, 2024-01-25:
> Correct, I added cloudflare as an extra layer of protection but after enabling the authentication with google the mobileraker app does not connect anymore.
>
> When I open the page on a browser it will [...]

Alright,
that is currently not a supported use case.
However, if basic auth or another header based auth is available via cloudflare you can simply add the cloudflare tunnel as a manual remote connection and add the specific http header values.