Clon1998/mobileraker #328: Android monochrome icons

tui2019, 2024-01-26:
Monochrome icons were added in #88 but then removed in one of the updates. Is there a reason for this, or is it just a mistake?

Clon1998, 2024-01-26:
No real reason.
I assume that the plugin I use to auto-generate the icons removed them :)

Clon1998, 2024-01-26:
I just checked the repo and the monochrome icons are still present:

https://github.com/AustinSchulz/mobileraker/blob/dev/android/app/src/main/res/mipmap-anydpi-v26/ic_launcher.xml

https://github.com/AustinSchulz/mobileraker/blob/master/android/app/src/main/res/drawable-mdpi/ic_launcher_monochrome.png

tui2019, 2024-01-28:
As I understand, that is not a main repo, but rather a fork with monochrome icons.

Clon1998, 2024-01-28:
Ohh you are right. 
I did not notice that I was not on my repo lol. I will add them again.

commit 5177d1c7 (2024-01-28): Fix #328 - Added monochrome option again
closed 2024-01-28 by tui2019