Clon1998/mobileraker #330: Arrange printers' list / hide some printers

wikolii, 2024-01-28:
## Feature Request

### Problem Description
The printers' list keep changing the order. I can only guess it's random.

### Proposed Solution
I think that adding filters and sorting the list would be very helpful when using a large number of machines.
The main filter I'd be interested in is "hide offline" for example.
Also sorting alphabeticaly would be nice.

### Alternatives Considered
I'm currently traveling with 3 machines and I've considered removing the ones I didn't bring with me to make the list shorter and easier to switch between them. But I'm not certain if I'll need to setup everything again when I re-add the ones left behind (macros, and other settings)

Clon1998, 2024-01-28:
Hey,
what are you referring to when talking about the printer list? Are you talking about the overview page or about the dialog to switch between printers?

I probably don't own enough printers to experience the random ordering.

wikolii, 2024-01-28:
I don't recall adding them on this order.
The list's order is not actually very important, but being able to filter out offline instances at least, would be cool, thanks 

[image]
[image]
[image]

wikolii, 2024-04-08:
Turns out the list respects the order the printers were added on the first place. Thanks anyways!

Clon1998, 2024-07-03:
2.7.0 added the option to reorder printers via the "Overview Page".

closed 2024-07-03 by Clon1998

wikolii, 2024-07-12:
> 2.7.0 added the option to reorder printers via the "Overview Page".

Note: when you swipe left or right on the printer name to change from one device to another, it still follows the original order the printers were added, not the reorder.

wikolii, 2024-07-12:
https://github.com/user-attachments/assets/a7a2c686-ff10-4234-98a1-6438e6a2287e

I was swiping right every time

Clon1998, 2024-07-12:
Hey,
good to know. There is a possibility that the swipe action is using the wrong source and therefore is not ordered.

commit 7731e4cc (2024-07-13): Fix #330 - Ensured Switch Printer quick action honors order