Clon1998/mobileraker #334: Initiates Load filament in a middle of a print

Jigit3D, 2024-02-10:
## Bug Report

### Description
While printer is busy printing a model pushing “Load filament” will initiate filament loading. After load is finished, making a big bulb on a part, printer will continue to print.

### Steps to Reproduce
1. Start print job
2. Click on “Load filament” macro
4. See the error.

### Expected Behavior
After print job is initiated no other interactions with printer shall be available, except:
Cancel
Pause
Speed change
Flow rate change
Micro lifting adjustments
Temperature change extruded and table

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): iOS
- Mobileraker: 2.6.10-383
- Klipper: 0.12.0-102-g9f41f53c
- Moonraker: 0.8.0-308

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-02-10:
Hey,
There is no native option to load filament offered by mobileraker.
There is an extruder card that allows users to extrude the filament manually. However, this card is not shown while printing.

Could you provide me a screenshot with the load filament button you are referring to?

Clon1998, 2024-02-23:
Closed as info is missing and the app itself does not offer any load filament option.