Clon1998/mobileraker #337: Error while fetching printer data

dverdep, 2024-02-29:
## Bug Report

### Description
I am in the process of creating a custom printer, when I open the app it throws this error
Exception:
 MobilerakerException{Missing field: print, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:common/data/dto/machine/printer.dart:88)
#1      PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:212)
<asynchronous suspension>

I have seen others bring this issue up about having more than one extruder. I only have one initialized at a time but there are 2 other extruder_steppers that can be changed out for the main extruder (3-in 1-out). I saw someone got it fixed by downloading the beta version however I got the same error on both.

### Steps to Reproduce
1. Set up a printer as above
2. Add it to Mobileraker
3. View the dashboard
4. See the error

### Expected Behavior
I expect the interface to show the printer controls

### Screenshots
[image]

### Version Information

- Device-OS (Android/iOS): iPhone 12[iOS 17.3.1]
- Mobileraker: [2.6.12-404]
- Klipper: [v0.12.0-114-ga77d0790]
- Moonraker: [v0.8.0-318-g9813011d]

Clon1998, 2024-02-29:
> ## Bug Report
> ### Description
> I am in the process of creating a custom printer, when I open the app it throws this error Exception: MobilerakerException{Missing field: print, parentException: null, parentStack: null}
> 
> #0 PrinterBuilder.build (package:common/data/dto/machine/printer.dart:88) #1 PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:212)
> 
> I have seen others bring this issue up about having more than one extruder. I only have one initialized at a time but there are 2 other extruder_steppers that can be changed out for the main extruder (3-in 1-out). I saw someone got it fixed by downloading the beta version however I got the same error on both.
> 
> ### Steps to Reproduce
> 1. Set up a printer as above
> 2. Add it to Mobileraker
> 3. View the dashboard
> 4. See the error
> 
> ### Expected Behavior
> I expect the interface to show the printer controls
> 
> ### Screenshots
> ![error](https://private-user-images.githubusercontent.com/161671458/308782862-945337b9-eb63-4b20-b981-408e45ab72d8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxOTkwNTksIm5iZiI6MTcwOTE5ODc1OSwicGF0aCI6Ii8xNjE2NzE0NTgvMzA4NzgyODYyLTk0NTMzN2I5LWViNjMtNGIyMC1iOTgxLTQwOGU0NWFiNzJkOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOVQwOTI1NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZmFiNmMyNzRkMmU5ZWNmZGU5NTRhZWE0ZGVhMGFlZDllMGRjYWRjY2E3ZTczNzc4ODcwYWRjZjdiYWQxYjZmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.f6wlDFVuCpaZIZ4KI4JGqKnMirgctXG0u_tLbcUUvgk)
> 
> ### Version Information
> * Device-OS (Android/iOS): iPhone 12[iOS 17.3.1]
> * Mobileraker: [2.6.12-404]
> * Klipper: [v0.12.0-114-ga77d0790]
> * Moonraker: [v0.8.0-318-g9813011d]

Hey,
can you provide me your printer.cfg/klipper.log?
I believe the `print` status object, which contains data like print progress, current file.... is only available if you added the `virtual_sdcard` option to you config.

Clon1998, 2024-02-29:
[image]

https://moonraker.readthedocs.io/en/latest/printer_objects/#print_stats

dverdep, 2024-02-29:
yes you were correct i had not added the virtual sd yet and once added it booted right up. 
Thank you so much!!

closed 2024-02-29 by dverdep