Clon1998/mobileraker #341: Can not fetch supporter tiers!

smolz, 2024-03-09:
## Bug Report

### Description
After paying for the lifetime subscription, the app still asks me to become a supporter.

### Steps to Reproduce
Pay money.  
Close and reopen app
Hit the hamburger menu (top left) and select "Support the Dev!"
See output:
 Can not fetch supporter tiers!

### Expected Behavior
Recognize that I have paid

### Screenshots
[image]

### Version Information

- Device-OS (Android/iOS): iOS
- Mobileraker: 2.6.11 hotfix1
- Klipper: [version]
- Moonraker: [version]

### Debug Logs

### Additional Context

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-03-09:
Hey,
please reach out to me via mail (dev@mobileraker.com) or via discord (@pad_sch) as posting any transaction data that I might need is a bad idea.

Please provide me your mobileraker log via either channel

Clon1998, 2024-03-09:
Close as this was most likely caused by apples servers.

closed 2024-03-09 by Clon1998