Clon1998/mobileraker #345: When Klipper UI secured with basic auth camera stream not visible in the app

VivantSenior, 2024-03-16:
## Bug Report

### Description
When klipper is secured with basic auth and self signed certificate (as described [here](https://forum.vorondesign.com/threads/protect-your-klipper-printer-mainsail-password-ssl-firewall-rules.469/)) Mobileraker can't load camera image / stream. Mobileraker tries to open camera stream via https://klipper.local:7130/webcam/?action=snapshot but should use link without the port: https://klipper.local/webcam/?action=snapshot and ask for credentials. A way to store it in the app would be useful.

### Steps to Reproduce
1. Setup Klipper, mainsail, moonraker, mobileraker, crowsnest
2. Do some changes in moonraker.conf:
```
[server]
host: 0.0.0.0
port: 7125
ssl_port: 7130
klippy_uds_address: /home/user/printer_data/comms/klippy.sock
max_upload_size: 1024

[authorization]
enable_api_key: True
default_source: moonraker
```
3. Setup basic auth
4. Setup camera with crowsnest
5. Open Mobileraker app

### Expected Behavior
Camera stream visible in Mobileraker.

### Screenshots
[image]

### Version Information

- Device-OS (Android/iOS): iOS 17.4
- Mobileraker: v0.4.0-54-gc252974f / iOS app: 2.6.13
- Klipper: v0.12.0-125-gbfb71bc2
- Moonraker: v0.8.0-324-ga3e4dac5

### Debug Logs
```
💡 AdaptiveMjpegManager started - targFps: 15 - https://klipper.local:7130/webcam/?action=snapshot
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 404 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 404 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.
```
[mobileraker_logs.log](https://github.com/Clon1998/mobileraker/files/14625502/mobileraker_logs.log)

### Additional Context
N/A

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-03-21:
Hey,
this is already supported. Note that you can configure an absolute or relative URL for the webcam.
Just try to use `https://klipper.local/webcam/?action=snapshot` as webcam URL. The app should use the BasicAuth/Headers you defined for the printer.

VivantSenior, 2024-03-21:
@Clon1998 thank you for the response. Now I got 401 (unauthorized) error:

```
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status
code of 401 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 401 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
```

I didn't know if Mobileraker will ask for the credential so I logged in from the default browser and Safari. Link you provided is working in both browsers but video stream is no available in Mobileraker even after app restart.

Clon1998, 2024-03-21:
> @Clon1998 thank you for the response. Now I got 401 (unauthorized) error:
>
> ```
> MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status [...]

Mobileraker wont ask for credentials.
You will need to setup the credentials as BasicAuth header for the printer. 

https://www.debugbear.com/basic-auth-header-generator

VivantSenior, 2024-03-21:
Thank you, now it works :)

For the future references:
1. Mobileraker -> Cog icon -> Add camera as absolute URL, e.g. https://klipper.local/webcam/?action=stream
2. Generate password using this page: https://www.debugbear.com/basic-auth-header-generator
3. Set generated hash in: Mobileraker -> Cog icon -> Section HTTP-HEADERS -> Add header with name: "Authorization", header value: "Basic type_your_base64_hash_here" (excluding quotes)
4. Save

closed 2024-03-21 by VivantSenior