Clon1998/mobileraker #347: Mcu is off, can't access power controls

kazolar, 2024-03-21:
## Bug Report

### Description
I have power controls configured in moonraker, in fluid I can go to power and turn on the printer, then firmware restart with a short 3 sec delay, and everything works. 
I used to have access to power controls, but a recent update now grays out the controls.

Easy to reproduce. Use power controls (i.e tasmota) in Moonraker to turn the printer off, pi is still on, moonraker still runs, klipper screen allows for power to be turned on. Mobileraker used to, recent update took that out.

### Version Information

- Device-OS (Android/iOS): android
- Mobileraker: [version] March 15, 2024
- Klipper: [version] latest, v12
- Moonraker: [version], latest

Clon1998, 2024-03-21:
Hey,
this is a known bug which I already addressed in the dev build. I am planning on releasing a hot fix on Friday

kazolar, 2024-03-22:
only thing keeping me from paying for the app (i know can use it for free). If can then use it to access all of my klipper machines from one spot, turn them on, pre-heat from the phone, worth it to me -- I signed up for pay version of octoanywhere, and will gladly contribute here.

Clon1998, 2024-03-25:
Fixed in 2.6.14 (Rolling out now)

closed 2024-03-25 by Clon1998