Clon1998/mobileraker #349: WebRTC/MediaMTX Support

ihatemyisp, 2024-03-26:
## Feature Request

### Problem Description
I use a couple of Wyze cameras to monitor my printer and enclosure. They are streamed to Mainsail via WyzeBridge which uses WebRTC/MediaMTX. Support for MediaMTX is baked into Mailsail and it works flawlessly.

### Proposed Solution
Add WebRTC/MediaMTX support.

### Alternatives Considered
Unfortunately in my case, there is no alternative that makes sense.

### Additional Context
None necessary, I believe.

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [x] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2024-03-26:
Can you provide me an example code that shows how the messaging for MediaMTX works?

ihatemyisp, 2024-03-26:
I've actually been digging through it trying to figure out if I could get this working myself. I'm having issues making heads or tails of all of it, hah.

WyzeBridge gives me a WebRTC URL (i.e., http://3d2.lan:8889/sv06-plus/) and I plug that into Mainsail. I got started down this rabbit hole via the Crowsnest docs: https://crowsnest.mainsail.xyz/faq/how-to-use-wyze-cams.

When I load the URL by itself I get a player and the stream starts. Here's the source from that: https://url.vogon.dev/YellowGummyNorth

Now that I'm thinking about it, I kind of feel like I could've looked at the Mainsail source and tried to grab more info on how they did it before posting. My apologies.

ihatemyisp, 2024-04-23:
https://github.com/mainsail-crew/mainsail/blob/develop/src/components/webcams/streamers/WebrtcMediaMTX.vue

It looks very similar to their WebrtcGo2rtc implementation...I think.

kamermans, 2024-08-13:
@Clon1998 I'm also interested in this, and I'm happy to provide whatever you need 👍 

This PR implements MediaMTX in Fluidd and includes a docker container for testing MediaMTX locally:
https://github.com/fluidd-core/fluidd/pull/1409

Clon1998, 2024-08-14:
I will look into this once I am back home in two weeks

kamermans, 2024-08-16:
Thanks @Clon1998, I'm not a Dart developer, but I do know many other languages, so I'll take a look at it unless you beat me to it :)

commit fa78e2f9 (2024-08-26): Resolves #349 - Added MediaMTX manager
closed 2024-08-26 by Clon1998