Clon1998/mobileraker #350: Time remaining in notifications is wrong

spitfire, 2024-03-28:
[image]
The time in notifications is wrong. Live activities and mainsail show the correct one, printer's timezone is set up correctly.

Clon1998, 2024-03-28:
> ![image](https://private-user-images.githubusercontent.com/42872/317883521-01fe925e-0f7e-4c89-9680-69a39e68b2a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3Lmd [...]

Is the timezone also correct in the mobileraker.conf file?

spitfire, 2024-03-29:
It is not, but to make it stranger it is set to CST, which is 6 hours behind, and the notification's time is only 1 hour behind. I'll try to adjust it once the print is over, cause changing it would probably restart something and interrupt it;)

spitfire, 2024-03-29:
Adjusted it to CET, restarted Mobileraker companion (through mainsail) and it's still showing the same offset/timezone on the next print.

spitfire, 2024-03-29:
[image]
Now it is correct ¯\_(ツ)_/¯

closed 2024-04-26 by Clon1998