Clon1998/mobileraker #351: Macro Categories [FR]

jimmysdiy, 2024-03-29:
## Feature Request
Requesting for you to consider adding Macro Categories as oppose to cluttered up. This is similar to Mainsail Categorizing the Macros how you want.

### Problem Description
Not a major issue. However, this causes clutter on the screen and sometime requires more time to find a macro

### Proposed Solution
Implement Macro Categories

Clon1998, 2024-03-29:
Hey,
this is already supported since V0.1.0.
Open up the printer settings (Open the menu, Gear Icon at the top) and create a new group, add maros and save the changes.

closed 2024-03-29 by Clon1998

jimmysdiy, 2024-03-30:
Hi - I do have Macro's setup in Mainsail (see attached). However, the categories do not push over to the Mobileraker App (see attached).

The request is to push Categories over to the App. It will make life easy.

[image]
[image]

Clon1998, 2024-03-30:
There is currently no sync of macro groupa between Mobileraker and other UIs planned.

As I mentioned, you can define groups in the printer settings.

https://github.com/Clon1998/mobileraker/assets/10357775/55d5f744-cd3a-4030-8d4a-0c7d644d3830

jimmysdiy, 2024-03-30:
Odd that you have a drop down to select group of macros. I don't even get the drop down. You can clearly see that I have a grouping done in the mainsail. Is there an option that I am missing?

Clon1998, 2024-03-30:
> Odd that you have a drop down to select group of macros. I don't even get the drop down. You can clearly see that I have a grouping done in the mainsail. Is there an option that I am missing?
> 
> Nvn. Found it.

It is not weird at all.
Again, the settings are NOT synced between mainsail and mobileraker. 
You will need to define groups within the mobileraker if you want to use them.