Clon1998/mobileraker #352: Support klipper-for-cnc with XYZABCE Axes

noobydp, 2024-04-01:
## Feature Request
I use the fork "klipper-for-cnc" https://github.com/naikymen/klipper-for-cnc on my CNC router, but it has support for additional axes "XYZABCE". I'd like to use Mobileraker as a mobile client, but it throws errors when adding the machine.

### Problem Description
I believe the issue is caused by a change in the toolhead object which has additional values for the axes, even for machines with only XYZ movement axes (like mine)
The toolhead object includes null or numeric values for the ABC axes.

i.e. from /printer/objects/query?toolhead data
`{
    "result": {
        "eventtime": 71.727285754,
        "status": {
            "toolhead": {
                "homed_axes": "",
                "axis_minimum": [
                    0.0,
                    0.0,
                    0.0,
                    null,
                    null,
                    null,
                    null
                ],
                "axis_maximum": [
                    299.5,
                    180.0,
                    42.0,
                    null,
                    null,
                    null,
                    null
                ],
                "print_time": 0.0,
                "stalls": 0,
                "estimated_print_time": 75.15606992,
                "extruder": "extruder",
                "position": [
                    0.0,
                    0.0,
                    0.0,
                    **null,
                    null,
                    null,**
                    0.0
                ],
                "max_velocity": 20.0,
                "max_accel": 50.0,
                "max_accel_to_decel": 25.0,
                "square_corner_velocity": 5.0
            }
        }
    }
}`

### Proposed Solution
I'm not sure if the position data for the ABC axes is important (as I don't have them installed right now).
Extending the Mobileraker UI to show/support the additional axes sounds like a ton of work (although the ideal solution), but it would still be great if we could use Mobileraker and have it ignore the ABC axes.

### Alternatives Considered
Since I only use XYZ right, I considered requesting changes in the klipper-for-cnc fork to send a compliant object, however adding the ABC axes in the future would break the app again. I may add another axis soon so thought adding support here would be better long term.

### Additional Context
Errors from the app:
```
Exception:
 type 'Null' is not a subtype of type 'num' in type cast

#0      _$$GCodeMoveImplFromJson.<anonymous closure> (package:common/data/dto/machine/gcode_move.g.dart:23)
#1      MappedIterable.elementAt (dart:_internal/iterable.dart:385)
#2      ListIterator.moveNext (dart:_internal/iterable.dart:354)
#3      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189)
#4      new _GrowableList.of (dart:core-patch/growable_array.dart:150)
#5      new List.of (dart:core-patch/array_patch.dart:39)
#6      SetBase.toList (dart:collection/set.dart:119)
#7      _$$GCodeMoveImplFromJson (package:common/data/dto/machine/gcode_move.g.dart:24)
#8      new _$GCodeMoveImpl.fromJson (package:common/data/dto/machine/gcode_move.freezed.dart:229)
#9      _$GCodeMoveFromJson (package:common/data/dto/machine/gcode_move.freezed.dart:18)
#10     new GCodeMove.fromJson (package:common/data/dto/machine/gcode_move.dart:40)
#11     new GCodeMove.partialUpdate (package:common/data/dto/machine/gcode_move.dart:46)
#12     PrinterService._updateGCodeMove (package:common/service/moonraker/printer_service.dart:743)
#13     PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:587)
#14     PrinterService._temperatureStore.<anonymous closure> (package:common/service/moonraker/printer_service.dart:536)
#15     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:633)
#16     PrinterService._parseQueriedObjects (package:common/service/moonraker/printer_service.dart:694)
#17     PrinterService._printerObjectsQuery (package:common/service/moonraker/printer_service.dart:860)
<asynchronous suspension>
#18     PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:225)
<asynchronous suspension>

Failed-Key: gcode_move
```

```
Exception:
 MobilerakerException{Missing field: toolhead, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:common/data/dto/machine/printer.dart:90)
#1      PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:230)
<asynchronous suspension>
```
---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x ] Provided a clear description of the problem you're facing.
- [x ] Clearly outlined the proposed solution or feature.
- [x ] Mentioned any alternative solutions or ideas you've considered.
- [x ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2024-04-01:
Hey,
I will have a look to see if there is a possibility to update the parser of the different data objects. However, I think there will be more parsing errors after we fix this one.

Clon1998, 2024-04-01:
The error you posted above is actually caused by the `gcode_move` object (The toolhead will fail too tho).
Can you provide me the query response of the gcode_move `/printer/objects/query?gcode_move`?

noobydp, 2024-04-01:
Thank you, I'm happy to help/test, but my dev skills are pretty limited.

/printer/objects/query?gcode_move
```
{
    "result": {
        "eventtime": 41.543546703,
        "status": {
            "gcode_move": {
                "speed_factor": 1.0,
                "speed": 1500.0,
                "extrude_factor": 1.0,
                "absolute_coordinates": true,
                "absolute_extrude": true,
                "homing_origin": [
                    0.0,
                    0.0,
                    0.0,
                    null,
                    null,
                    null,
                    0.0
                ],
                "position": [
                    0.0,
                    0.0,
                    0.0,
                    null,
                    null,
                    null,
                    0.0
                ],
                "gcode_position": [
                    0.0,
                    0.0,
                    0.0,
                    null,
                    null,
                    null,
                    0.0
                ]
            }
        }
    }
}
```