Clon1998/mobileraker #354: Webcam panel shown despite no webcam configured

AndreKR, 2024-04-08:
## Bug Report

### Description
I don't have a webcam and there is no webcam configured in the printer settings.

Yet, since I updated Mobileraker (I know, my own fault) there is a webcam panel on the main screen saying "Invalid arguments".

### Steps to Reproduce
1. Make sure there is no webcam configured for a printer.
2. Open Mobileraker.
3. See the error.

### Expected Behavior
I expect there to be no webcam panel when there is no webcam available.

### Screenshots
[image]

[image]

### Version Information

- Device-OS (Android/iOS): Android
- Mobileraker: 2.6.14-424
- Klipper: v0.12.0-118-g18de421c
- Moonraker: v0.8.0-320-g6b1b8c5

### Debug Logs
[mobileraker_2024-04-08T02-12-37.log](https://github.com/Clon1998/mobileraker/files/14899564/mobileraker_2024-04-08T02-12-37.log)

Clon1998, 2024-04-08:
Thanks for reporting this bug.
I will double-check that, but I thought I had already fixed it. Maybe the fix is currently only on dev tho.

Clon1998, 2024-04-26:
Fixed in 2.6.15/2.6.16

closed 2024-04-26 by Clon1998