Clon1998/mobileraker #355: Add custom headers to Webcam URLs?

bertabus, 2024-04-09:
## Feature Request
Is there any way to pass the custom  headers with the webcam urls?

### Problem Description
I place my server behind a Caddy server with ssl and basic auth, this allows me to have SSL encryption via letsencrypt.
I can place my basic auth header in the custom headers. 
This allows me access to the klipper/moonraker services.
I don't however get webcam support, error 401, which usually means it's an auth error.

### Proposed Solution
Pass custom headers to the webcam url requests.
Is there a case when you wouldn't want the same custom headers passed for accessing other URL's?

### Alternatives Considered
Currently opened up my "/webcam*" url's under Caddy. This makes it unencrypted/unprotected. 
I'm less concerned with someone watching my printing than having access to it, so it works for now.

Clon1998, 2024-04-10:
I'll need to double check thet. But I think the webcams will use the same headers as the the printer as long as the base url/host is the same.

Clon1998, 2024-04-10:
> I'll need to double check thet. But I think the webcams will use the same headers as the the printer as long as the base url/host is the same.

Nvm. I thought I implemented it that way, but it seems like I did not. I will have a look at how hard it is to implement custom headers for webcams since the settings are synced between all UIs. In any case, the webcam should use the same headers as the printer.

Clon1998, 2024-04-28:
Hey,
just wanted to give you an update. In my original answer 3 weeks ago I mentioned that the webcams do not use the same headers as defined in the printer's settings. However, I was wrong about that and in fact they are using the same headers by using the same http client as the rest of the app.

Let me know if that helps or if you still would need to use custom headers due to the usage of webcams running on a different host etc.

Clon1998, 2024-05-23:
Closed due to inactivity

closed 2024-05-23 by Clon1998