Clon1998/mobileraker #359: app crashes on first attempt, loads normally 2nd attempt

Grafton1046, 2024-04-24:
## Bug Report

### Description
Frequently when launching mobile raker it flashes open and then closes out. Tapping the app a second allows it open successfully. As a result I'm getting occasional reminders that the app is crashing.

[mobileraker_2024-04-23T20-12-57.log](https://github.com/Clon1998/mobileraker/files/15095716/mobileraker_2024-04-23T20-12-57.log)
[mobileraker_2024-04-23T19-45-03.log](https://github.com/Clon1998/mobileraker/files/15095717/mobileraker_2024-04-23T19-45-03.log)
[mobileraker_2024-04-24T10-17-51.log](https://github.com/Clon1998/mobileraker/files/15095718/mobileraker_2024-04-24T10-17-51.log)
[mobileraker_2024-04-24T10-07-55.log](https://github.com/Clon1998/mobileraker/files/15095719/mobileraker_2024-04-24T10-07-55.log)
[mobileraker_2024-04-24T10-04-02.log](https://github.com/Clon1998/mobileraker/files/15095720/mobileraker_2024-04-24T10-04-02.log)

[image]

### Steps to Reproduce
Attempt to open app, issue is intermittent 

### Expected Behavior
App opens to overview panel

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): Android 14 on Samsung s21 Ultra
- Mobileraker: 2.6.14-424
- Klipper: v0.12.0-171
- Moonraker: v0.8.0-327

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-04-24:
Hey,
thanks for reporting that issue. Does this error happen occasionally or all the time?
Also, did you try to reset the app's storage yet?

Grafton1046, 2024-04-24:
> Hey, thanks for reporting that issue. Does this error happen occasionally or all the time? Also, did you try to reset the app's storage yet?

around once every 3 or so launches, did not, will do that and test.

Grafton1046, 2024-04-24:
> > > Hey, thanks for reporting that issue. Does this error happen occasionally or all the time? Also, did you try to reset the app's storage yet?
> > 
> > 
> > around once every 3 or so launches, did not, will do that and test.
> 
> Thanks! I will look into that. Sadly, logging starts way later and therfore does not provide any useful info. Can you do me a favor and enable crashalytics reporting in the app settings? It should catch the error and report it to me :).

that was turned on already

Clon1998, 2024-04-24:
Just pushed 2.6.15 to play store. 
Let me know if it helps with that issue!

Clon1998, 2024-04-26:
I just build another version ([2.6.16](https://github.com/Clon1998/mobileraker/releases/tag/android2.6.16)) that might resolve the issue. You can download it from github or wait for Google Play to be ready, as downloading from github tends to break the auto-update from Google play.

Turge08, 2024-04-28:
I'm on 2.6.16 and I have what I assume is the same issue. It seems like it happens when trying to launch the app after a period of inacvity.

On my S23 Ultra, if I click on "Optimize Now" in Device Care, it shows that it killed 1 background process. When I try to launch mobileraker afterwards, it crashes immediately. I do have mobileraker set to "Unrestricted" for battery/background use.

Here's the logcat error:

`04-28 16:02:18.801 20885 20885 E AndroidRuntime: FATAL EXCEPTION: main
04-28 16:02:18.801 20885 20885 E AndroidRuntime: Process: com.mobileraker.android, PID: 20885
04-28 16:02:18.801 20885 20885 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.mobileraker.android/com.mobileraker.MainActivity}: java.lang.IllegalStateException: Default FirebaseApp is not initialized in this process com.mobileraker.android. Make sure to call FirebaseApp.initializeApp(Context) first.
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:5427)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:5508)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:57)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:180)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:98)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2685)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:106)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.os.Looper.loopOnce(Looper.java:230)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:319)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:8919)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:578)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
04-28 16:02:18.801 20885 20885 E AndroidRuntime: Caused by: java.lang.IllegalStateException: Default FirebaseApp is not initialized in this process com.mobileraker.android. Make sure to call FirebaseApp.initializeApp(Context) first.
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at r8.g.o(Unknown Source:56)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at y9.b.d(Unknown Source:3)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at ae.h.n(Unknown Source:21)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at ae.h.g(Unknown Source:1)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at io.flutter.embedding.engine.c$c.i(Unknown Source:18)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at io.flutter.embedding.engine.c.g(Unknown Source:14)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at io.flutter.embedding.android.d.v(Unknown Source:20)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at io.flutter.embedding.android.c.onNewIntent(Unknown Source:13)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.Activity.performNewIntent(Activity.java:8993)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1556)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1569)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4362)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:5392)
04-28 16:02:18.801 20885 20885 E AndroidRuntime:        ... 13 more`

Hope this helps to resolve the error.

Clon1998, 2024-04-28:
> I'm on 2.6.16 and I have what I assume is the same issue. It seems like it happens when trying to launch the app after a period of inacvity.
>
> On my S23 Ultra, if I click on "Optimize Now" in Device [...]

Thanks!
I also saw that it still occurs on 2.6.16. There is not much I can do as this seems to be caused by the firebase plugin itself. 
Worst case, I'll need to remove it and write some UI stuff for the firebase login myself...

Turge08, 2024-04-28:
It's been a while since I've developed for Android but perhaps this might help

https://wajahatkarim.com/2020/03/defending-your-app/

Clon1998, 2024-05-07:
Hey,
I just started the rollout for the preview/beta of 2.6.19 (android2.6.19)

I removed the dependency that caused this issue entirely. Fell free to download the latest RC vis GitHub or via the Playstore once available and let me know if the issue persists.
On my end I'll monitor the vitals provided by Google Play to see if the crash still occurs too.

Clon1998, 2024-05-10:
I'll close this issue as it seems to be resolved.

I slowly rolled out 2.6.19 and monitored the crash reports and can say with confidents that it is fixed as I currently have around 2000 installs without a single error report in google play 🥳

closed 2024-05-10 by Clon1998