Clon1998/mobileraker #360: Neopixel (Possibly all) LED color wheel bugged beyond use

RebelP72, 2024-04-26:
## Bug Report

### Description
A clear and concise description of the bug.

### Steps to Reproduce
1. Go to second page 
2. Click on a led wheel
4. See the error.

### Expected Behavior
functional color wheel 

### Version Information

IOS
2.6.15-441
0.12.0-85
0.8.0-268

I have reinstalled the app and companion

--

https://github.com/Clon1998/mobileraker/assets/133155700/bdd6449c-f567-47b4-a9c6-ef78b015e7d4

Clon1998, 2024-04-26:
thanks for letting me know.
Seems like the color wheel dependency is broken.

Clon1998, 2024-04-26:
I had a look at that issue and sadly this seems to be a flaky one.
The picker is working on the emulator for Android as well as on an emulated IPhone 15 Pro & IPhone 13.
However, I am seeing the same behavior on my personal IPhone 13.

What iPhone + iOS version are you using?

Clon1998, 2024-04-26:
Related to https://github.com/mchome/flutter_colorpicker/issues/104

closed 2024-04-26 by Clon1998

Clon1998, 2024-04-26:
Version 2.6.16 is awaiting approval by apple and afterwards rolling out during slowly.

Let me know if the issue is resolved on your device too.

RebelP72, 2024-04-26:
Sorry for not replying during the troubleshooting posted and went to sleep as soon as the update hits my phone will update

RebelP72, 2024-04-26:
Updated:
Middle part works however unable to verify if just me but outer slider doesn't work

Attempt to rotate to different color just slides down https://github.com/Clon1998/mobileraker/assets/133155700/1704078b-75fc-4969-ba7b-27bd35027022

Clon1998, 2024-04-26:
> Updated: Middle part works however unable to verify if just me but outer slider doesn't work
> 
> Attempt to rotate to different color just slides down https://github.com/Clon1998/mobileraker/assets/133155700/1704078b-75fc-4969-ba7b-27bd35027022
> 
> Edit: Restarted phone issue appears fixed

Yeah the outer slider is a bit finicky due to the dialog being scrollable. I'll need to look into that and find a solution...

commit ee63b1a1 (2024-04-26): Fix #360