Clon1998/mobileraker #362: Screw tilt adjust popup on opening app

Johnnytk36, 2024-04-27:
## Bug Report

### Description
Since the last two updates this week when I open the app I randomly get a tilt screw adjust pop up. 

### Steps to Reproduce
1. Open app
2. Exit app
3. Wait a couple minutes
4. Go back into app and when it resumes i get a tilt screw adjust popup

### Expected Behavior
No till screw adjust pop up unless I request it. 

### Screenshots
[image]

### Version Information

- Device-OS (Android/iOS): [iOS]
- Mobileraker: [latest]
Latest sonic pad version. 

### Debug [Logs]
[mobileraker_2024-04-27T16-10-22.log](https://github.com/Clon1998/mobileraker/files/15140025/mobileraker_2024-04-27T16-10-22.log)
[mobileraker_2024-04-27T14-40-59.log](https://github.com/Clon1998/mobileraker/files/15140026/mobileraker_2024-04-27T14-40-59.log)
[mobileraker_2024-04-26T20-09-52.log](https://github.com/Clon1998/mobileraker/files/15140027/mobileraker_2024-04-26T20-09-52.log)
[text.txt](https://github.com/Clon1998/mobileraker/files/15140028/text.txt)
[mobileraker_2024-04-26T20-34-29.log](https://github.com/Clon1998/mobileraker/files/15140029/mobileraker_2024-04-26T20-34-29.log)
[mobileraker_2024-04-27T15-03-46.log](https://github.com/Clon1998/mobileraker/files/15140030/mobileraker_2024-04-27T15-03-46.log)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

igl-x2, 2024-04-27:
same here with lastest Android version .16
[mobileraker_2024-04-27T23-06-17.log](https://github.com/Clon1998/mobileraker/files/15140065/mobileraker_2024-04-27T23-06-17.log)

Johnnytk36, 2024-04-27:
Video of it happening. [https://share.icloud.com/photos/0b4B08Nz9fZcHZFSonYZ6ZuKQ](url)

Clon1998, 2024-04-27:
Hey,
I have noticed the same thing and already pushed a fix for that behavior. The dialog tents to open to often after the app was in the background for to long and caused a refresh of printer data.
I already pushued a fix that should mitigate that behavior which I will include as part of 2.6.17 next week. I halted the rollouts of 2.6.15 and 2.6.16 for both stores now which I i initially pushed due to a majore issue on Android that prevented the app from starting in some cases.

Also the printer I used to test the dialog and triggering isn't my primary one so I never noticed that issue until pushing patch 16...

closed 2024-04-29 by Clon1998