Clon1998/mobileraker #363: Reverse Proxy Camera Error

fqscfqj, 2024-04-28:
## Bug Report

### Description
I have reverse proxied my Fluidd and Moonraker into the network. When I enter the webpage, I can see the camera, but the camera does not display from within the app.

### Expected Behavior
Can display the camera like in a web page

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): [Android]
- Mobileraker: [2.6.16]
- Klipper: [v0.12.0-175]
- Moonraker: [v0.8.0-337]

### Debug Logs
[
[mobileraker_2024-04-28T23-39-18.log](https://github.com/Clon1998/mobileraker/files/15142643/mobileraker_2024-04-28T23-39-18.log)
](url)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

fqscfqj, 2024-04-28:
[image]
[image]

Clon1998, 2024-04-28:
Hey,
it seems like the webcam resolves to an unvaild URL. 
What happens if you try to enter the snapshotURI displayed in the app within your browser?

Can you compare it to the URL that is used within mainsail?

fqscfqj, 2024-04-29:
https://mo.xxxx.xyz:4443/ is the reverse proxy address for Moonraker.
https://voron.xxxx.xyz:4443/ is the reverse proxy address for Fluidd.
Therefore, the camera at the mo address should be inaccessible. The correct link is https://voron.xxxx.xyz:4443/webcam/?action=stream&cacheBust=1714353482963

fqscfqj, 2024-04-29:
Okay, I understand now, it's because Fluidd was not added to mobileraker.

closed 2024-04-29 by fqscfqj