Clon1998/mobileraker #364: Spoolman Rounding

brando132, 2024-04-28:
## Feature Request

Spoolman spool info card uses poor formatting above 1kg. instead of a gram unit display (451/1000grams), it uses kilograms when loaded with spools above 1kg. (3kg / 5kg)

### Proposed Solution
Change the default formatting to display in grams, or increase the number of sigfigs. 
3kg / 5kg  --> 3.141kg / 5kg or 3141g / 5000g

### Alternatives Considered
Potentially enable this as a toggle in app settings.

spitfire, 2024-05-23:
[image]
It is showing like this on a 1100g spool (Jayo makes 1,1kg ones) where I've used ~50g.

Clon1998, 2024-05-23:
Hey,
I will update the implementation to actual show a decimal point for kg too or as suggested make it configurable.

Clon1998, 2024-05-23:
Fixed 
[image]

spitfire, 2024-05-23:
Yessss. Accuracy to the second decimal (no more, and no less) is exactly what we needed :)

Clon1998, 2024-06-25:
Added in 2.7.0.
Just triggered the build. Rollout should start tomorrow und ramp up slowly over the next 7 days.

closed 2024-06-25 by Clon1998