Clon1998/mobileraker #368: Android app can't connect to pinter after a few days

caste1200, 2024-05-17:
## Bug Report

### Description
Mobileraker stops connecting to my printer after a couple of days. I keep getting notifications of printing progress but can't connect on the app. 

### Steps to Reproduce
1. Link printer to mobileraker using tailscale
2. Keep using it for a couple of days even outside of local network
3. Stops working after a couple of days out of nowhere

### Expected Behavior
Being able to connect

### Screenshots
[image]
[image]

### Version Information

- Device-OS (Android/iOS): android 13
- Mobileraker: 2.6.19
- Klipper: latest to date
- Moonraker: latest to date

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ x] Provided a clear bug description.
- [ x] Listed detailed steps to reproduce the issue.
- [x ] Described the expected behavior.
- [ x] Included the Mobileraker version you are using.
- [x ] Attached Mobileraker's debug log files.
- [ x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-05-17:
Hey,
first of all, it seems like your printer and websocket address are different ignoring  '/websocket'. This should nearly never be the case.

Secondly, if the issue occurs, are other UIs working/is the printer still reachable?

caste1200, 2024-05-17:
> Hey, first of all, it seems like your printer and websocket address are different ignoring '/websocket'. This should nearly never be the case.
> 
> Secondly, if the issue occurs, are other UIs working/is the printer still reachable?

Hi, I can still reach the printer with Obico without any issues. And for the websocket, wich address should I use? When I enter on both my tailscale address, it is not able to connect

Clon1998, 2024-05-17:
I've just double checked and it actually does not matter to much what you enter there.

My current suspicion is that the tailscale network closes the connection/can be slow form time to time. Therefore try to increase the client timeout to 10/20sec to ensure the app has more time to fetch the data.

This can be done in the printer settings.

caste1200, 2024-05-17:
> I've just double checked and it actually does not matter to much what you enter there.
>
> My current suspicion is that the tailscale network closes the connection/can be slow form time to time. [...]

I tried with 60s and still same error

caste1200, 2024-05-17:
[image]
I just sent a temp tower and as you can see I have notifications on this but if I open the app, same error. Obico has no issues.
Maybe I did the configuration wrong? Is there somewhere a guide on how to connect the printer via tailscale?

Clon1998, 2024-05-18:
I'll get back to you tomorrow.
If I don't feel free to ping me again.

In the meantime, can you try to use the tail scale address in your browser to verify if that is also not working if mobileraker is not working?

Clon1998, 2024-05-18:
Hey,

I just checked and my remote connection is working, but I'm not using Tailscale. It's important to note that the notifications work independently of any remote connection, as they use Google's/Apple's push service servers to deliver notifications.

If the app stops connecting to the Tailscale instance again, please check if the issue also occurs in your browser. This would indicate a problem with your Tailscale setup.

caste1200, 2024-05-27:
its working again, restarted my phone, disabled and re-enabled tailscale on phone.
Also removed the key expiry... not sure if that is what it was causing it since I have the feeling that everytime it works, it only works for a couple of days and then stops working

closed 2024-07-03 by Clon1998