Clon1998/mobileraker #370: Ercf / happy hare Support

strobobaer, 2024-05-21:
## Feature Request

### Problem Description
A clear and concise description of the problem or pain point you're experiencing. For example: "I'm always frustrated when..."

### Proposed Solution
A clear and concise description of the feature you'd like to see implemented.

### Alternatives Considered
If you've thought of any alternative solutions or features, please briefly describe them here.

### Additional Context
Add any screenshots, examples, or additional context to help us understand your feature request better.

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ ] Provided a clear description of the problem you're facing.
- [ ] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2024-05-21:
Hey,
what exactly is required for Happy Hare?
I assume some additional UI components?

strobobaer, 2024-05-22:
I Like a UI Look likes Klipperscreen „happy hare Edition“.Did u understand.Von meinem iPhone gesendet

Clon1998, 2024-05-22:
Sadly I sold my ERCF 
And therefore have no Idea how the UI is supposed to look or what functionalities are required.

Can you provide me some examples or screenshots?

strobobaer, 2024-05-22:
https://github.com/moggieuk/KlipperScreen-Happy-Hare-Edition

I Think this is the best Dokumentation

husnainali09, 2025-01-21:
Any updates on this ? I keep getting filament sensor errors on mobileraker

Clon1998, 2025-01-21:
> Any updates on this ? I keep getting filament sensor errors on mobileraker

can you specify what kind of errors?

pbutler, 2025-02-27:
I think what he means (and what I see).  Is that with ERCF with toolhead and extruder sensors are sending alerts when the filament switches out.  The sensors are being triggered but it's part of normal operation.

Clon1998, 2025-02-27:
@husnainali09 
If youre talking about the sensors. You can always disable them via the config file:
https://github.com/Clon1998/mobileraker_companion?tab=readme-ov-file#companion---config
```
ignore_filament_sensors:
# Comma separated list of filament sensors to ignore, ignored filament sensors do not trigger 
# a notification if they are triggered. This is useful if you have a filament sensor that is used
# in a MMU setup like ERCF.
# IMPORTANT, do not include the sensor type. E.g. if your sensor is configured
# like: [filament_switch_sensor printhead_sensor] add `printhead_sensor` to the list.
# Default: empty
# Optional
```

pbutler, 2025-02-27:
Ah thanks thats good to know.  I guess it helps to RTFM :)