Clon1998/mobileraker #372: Support the Dev page not working (only tested on IOS)

wikolii, 2024-06-03:
## Bug Report

### Description
Supporter tiers on the IOS app are unfetchable 

### Steps to Reproduce
1. Go to 'Menu'
2. Click on 'Support the Dev!'
3. See the error.

### Expected Behavior
I was trying to see the plans / prices available.

### Screenshots
[image]

### Version Information

- iOS
- Mobileraker: 2.6.19-468
- Klipper:  v0.12.0-155
- Moonraker: v0.8.0-325

### Debug Logs
[mobileraker_2024-05-31T00-30-57.log](https://github.com/user-attachments/files/15534237/mobileraker_2024-05-31T00-30-57.log)
[mobileraker_2024-06-01T17-41-23.log](https://github.com/user-attachments/files/15534239/mobileraker_2024-06-01T17-41-23.log)
[mobileraker_2024-06-01T20-54-27.log](https://github.com/user-attachments/files/15534240/mobileraker_2024-06-01T20-54-27.log)
[mobileraker_2024-05-31T13-06-38.log](https://github.com/user-attachments/files/15534241/mobileraker_2024-05-31T13-06-38.log)
[mobileraker_2024-06-03T00-17-53.log](https://github.com/user-attachments/files/15534242/mobileraker_2024-06-03T00-17-53.log)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- [X] Attached Mobileraker's debug log files.
- [X] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-06-03:
> Hi, it's been like this for a couple of days. I added an 11th printer to the farm and i wanted to check if the price range was acceptable. Thanks!

Hey,
this issue is caused by the fact that my apple developer membership expired and did not auto renew, nor was I able to renew it manually.
I am already in contact with apple to resolve this issue. 
Currently I can not give you any timeline as the issue is on apple's end.

If you're just looking for pricing :
The price for the supporter tier is about 2€/Month or 33€ life time (Might be slightly different due to local currency).

However, without an active developer license on my end I can not process any payments.

wikolii, 2024-06-03:
Hi there.

Ok, thanks. I will keep trying in the following days

closed 2024-06-03 by wikolii