Clon1998/mobileraker #373: Yggdrasil support

the-nexi, 2024-06-05:
## Feature Request

### Problem Description
It is impossible to connect via yggdrasil network, due to URL validation. 
### Proposed Solution
If yggdrasil addresses would be considered as valid, this would be greatly appriciated
### Alternatives Considered
It is impossible to provide an alternative solution, since yggdrasil addresses can't be validated by the app

### Additional Context
[image]

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [X] Provided a clear description of the problem you're facing.
- [X] Clearly outlined the proposed solution or feature.
- [X] Mentioned any alternative solutions or ideas you've considered.
- [X] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2024-06-05:
Hey,
that's already supported indirectly.
Just enter any IP to add the printer.
Afterwards, you can modify the address to whatever format you desire through the printer edit page which does not impose any format restrictions.

the-nexi, 2024-06-05:
> Hey,
> that's already supported indirectly.
> Just enter any IP to add the printer.
> Afterwards, you can modify the address to whatever format you desire through the printer edit page which does not impose any format restrictions.

I'm sorry, but it looks like like format restrictions are imposed anyways:

[image]

Clon1998, 2024-06-05:
You are right.
I will have a look :)

the-nexi, 2024-06-05:
> You are right.
> I will have a look :)

Thanks :3

Also when working on it, please make sure yggdrasil addresses can be also used in «Configure remote connection» > «Manual»