Clon1998/mobileraker #374: Choppy Video/GUI - Android

fuzelet, 2024-06-13:
## Bug Report

### Description
I have noticed that while using the Android version of the application, it stuggles to show 15fps webcams and the overall GUI seems rather choppy.  When viewing the exact same printer in the iOS application, it is easily able to render 15fps webcams and the GUI seems snappier.  Is there a framerate limiter or something in the Android version as it relates to overall app responsiveness?  I stuggle to view 15fps streams in the android version without constant hiccups and fps drops even down to 0.01fps, where as I am not seeing the same on the iOS application.

### Steps to Reproduce
1. Add Printer
2. Add Webcam
4. Android App becomes super choppy over the course of a print or after opening/reopening/using phone, requiring closing/reopening, etc multiple times over
4a. iOS App does not appear to behave in the same manner and is displaying webcam and printer fine. can even display 30fps webcam streams fine

### Expected Behavior
Android application to be able to view 15fps streams, even 30 fps streams without the application hitching or "lagging"

### Screenshots
added video below showing "recording" of mobileraker today on my Pixel 8.

### Version Information

- Device-OS (Android/iOS): Android 14 Pixel 8 & iOS 17.5.1 iPhone 14
- Mobileraker: 2.6.21-490
- Klipper: v0.12.0-229-g589bd64c
- Moonraker: v0.8.0-379-g346a3d7c

### Debug Logs
[logs.zip](https://github.com/user-attachments/files/15826365/logs.zip)

### Additional Context
Video of choppyness today from Android App
https://drive.google.com/file/d/19pqqV1h82oi0XfLEpTH5LMD_Y8mHBRLP/view?usp=drive_link

Found a similar mention a while back on Android here
https://github.com/Clon1998/mobileraker/issues/29
---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ X ] Provided a clear bug description.
- [ X ] Listed detailed steps to reproduce the issue.
- [ X ] Described the expected behavior.
- [ X ] Included the Mobileraker version you are using.
- [ X ] Attached Mobileraker's debug log files.
- [ X ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-06-13:
Hey,
what kind of webcam are you using (Mjpeg, Adaptiv Mjpeg, WebRtc...)?
Also what is the configured FPS and Resolution of that cam within crownsnest?

The provided GDrive link is also locked/private.

fuzelet, 2024-06-13:
Hey! Webcam is set to MJPEG-Streamer.  It is 15 fps currently in crowsnest and set to 1920x1080.  I have also tried 15 fps with1280x720, 30 fps and 1280x720, 30fps and 1080, etc.

The link permissions should be fixed now.  Sorry about that.

https://drive.google.com/file/d/19pqqV1h82oi0XfLEpTH5LMD_Y8mHBRLP/view?usp=sharing

fuzelet, 2024-06-13:
here is another video on android of the printer "homing"
https://drive.google.com/file/d/1AGqfwvDqaxfeFFi7c_4cKfOaFNZCHna5/view?usp=sharing

here is the same printer doing the same action on iOS
https://drive.google.com/file/d/1vNORX1Q4MNxoG-P6CktkmNOP3gBmPi8n/view?usp=sharing

I understand there will be some some choppiness given wifi, video compression, 15/30 fps, etc but im hoping these comparisons show what I am least experiencing currently on Android and how it seems to be choppier then the iOS counterpart

Clon1998, 2024-06-14:
> here is another video on android of the printer "homing" https://drive.google.com/file/d/1AGqfwvDqaxfeFFi7c_4cKfOaFNZCHna5/view?usp=sharing
>
> here is the same printer doing the same action on iOS [...]

Thanks!
I will have a look to see if I can reproduce that issue. My assumption is that on the native end Android takes longer to process the MJPEG stream. Therefore, can you try the "Adaptive Mjpeg," which just queries screenshots of the webcam to see if that helps?

fuzelet, 2024-06-14:
Thank you for the suggestion.  I messed with the webcam and changed to adaptive mpjeg and tried various settings such as 30fps varible, 15 fps, but the Android version seems to struggle.  I included another video below showing android is only doing 5/6 fps on adaptive 15fps webcam.

https://drive.google.com/file/d/1A_n3_JUcabr4kP4R08IxhUFQmF-bdDRV/view?usp=sharing

I did adjust the camera to 720 and restarted crowsnest, however android was still only showing 6/7 fps even on 720, so i dont think thats the issue. I can take a video of that as well if you need to see it,

[crowsnest]
log_path: /home/ubuntu/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: ustreamer
enable_rtsp: false
rtsp_port: 8554
port: 8080
device: /dev/v4l/by-path/platform-fc880000.usb-usb-0:1.2:1.0-video-index0
resolution: 1920x1080
max_fps: 15
v4l2ctl: auto_exposure=3,exposure_dynamic_framerate=0,focus_automatic_continuous=0,power_line_frequency=2,focus_absolute=325

[cam 2]
mode: ustreamer
enable_rtsp: false
rtsp_port: 8555
port: 8081
device: /dev/v4l/by-path/platform-fc880000.usb-usbv2-0:1.1:1.0-video-index0
resolution: 1920x1080
max_fps: 15
v4l2ctl: auto_exposure=3,exposure_dynamic_framerate=0,focus_automatic_continuous=0,power_line_frequency=2,focus_absolute=250

Clon1998, 2024-07-03:
> Thank you for the suggestion. I messed with the webcam and changed to adaptive mjpeg and tried various settings such as 30fps varible, 15 fps, but it still seems to struggle. I included another video below showing android is only doing 5/6 fps on adaptive 15fps webcam.
> 
> https://drive.google.com/file/d/1A_n3_JUcabr4kP4R08IxhUFQmF-bdDRV/view?usp=sharing
> 
> I did adjust the camera to 720 and restarted crowsnest, however android was still only showing 6/7 fps even on 720, so i dont think thats the issue. I can take a video of that as well if you need to see it,
> 
> [crowsnest] log_path: /home/ubuntu/printer_data/logs/crowsnest.log log_level: verbose # Valid Options are quiet/verbose/debug delete_log: false # Deletes log on every restart, if set to true no_proxy: false
> 
> [cam 1] mode: ustreamer enable_rtsp: false rtsp_port: 8554 port: 8080 device: /dev/v4l/by-path/platform-fc880000.usb-usb-0:1.2:1.0-video-index0 resolution: 1920x1080 max_fps: 15 v4l2ctl: auto_exposure=3,exposure_dynamic_framerate=0,focus_automatic_continuous=0,power_line_frequency=2,focus_absolute=325
> 
> [cam 2] mode: ustreamer enable_rtsp: false rtsp_port: 8555 port: 8081 device: /dev/v4l/by-path/platform-fc880000.usb-usbv2-0:1.1:1.0-video-index0 resolution: 1920x1080 max_fps: 15 v4l2ctl: auto_exposure=3,exposure_dynamic_framerate=0,focus_automatic_continuous=0,power_line_frequency=2,focus_absolute=250

Sorry for my late reply, 1920x1080 can be rather demanding. Also the Pi itself tent to struggle (Just open your webinterface multiple times and you will notice that the FPS drop.).
I will try to see if I can improve the adaptive implementation of MJPEG but if the PI/Network is to slow I can't do much about it right now.

thePunderWoman, 2025-01-19:
I've been seeing this exact problem, too. I have a 3DO camera that in every other UI is running at 60fps, but in the Android app is extremely choppy. The printer and streamer are both on ethernet on the same network as my phone. I've been trying to figure out why, but seems like maybe it's an app issue given others are seeing it too. I'll install the iOS version on my iPad and see how that looks.

Clon1998, 2025-01-19:
> I've been seeing this exact problem, too. I have a 3DO camera that in every other UI is running at 60fps, but in the Android app is extremely choppy. The printer and streamer are both on ethernet on [...]

Okay!
Please let me know what you observe. I will also verify that another time but did not notice anything that stood out.