Clon1998/mobileraker #375: Printer unreachable on printer drop down list

wikolii, 2024-06-20:
## Bug Report

### Description
I recently paid the subscription to be able to add more printers.
I found that in my case there is no way to select the last printer on the list 

### Steps to Reproduce
1. Go to any printer specific page (dashboard, console, etc)
2. Click on the printer name on the top of the screen
3. Scroll down to the last printer
4. See the error. It can't be selected.

### Expected Behavior
Sleect the printer and show it's content

### Screenshots
[image]
### Version Information

- Device-OS (Android/iOS): iOS app v2.6.20-486
- Mobileraker: v0.4.0-63-gea7277ee
- Klipper: v0.12.0-229-g589bd64c
- Moonraker: v0.8.0-379-g346a3d7c

### Debug Logs

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-06-20:
Hey, thanks for reporting that issue.
I noticed that issue myself about two weeks ago and fixed that in f7c963d3dcd03f04b8050aaed2e0a44a9063046c.

If you want, you can join the [TestFlight](https://testflight.apple.com/join/h9JwuyR0) version, which already contains the fix for this issue alongside some other stuff ([Dev Changelog](https://github.com/Clon1998/mobileraker/blob/dev/docs/changelog.md)).

wikolii, 2024-06-21:
Thanks, loving these new updates

closed 2024-06-21 by wikolii