Clon1998/mobileraker #377: Add Tachometer Fan Reporting

TheMasterOfTech, 2024-06-24:
On the information card for fans, if a tachometer fan is utilized, display the RPM reading as reported by the fan’s tachometer in a small, but readable font size underneath the fan speed for a quick and efficient way to check fan speed on the fly.

Clon1998, 2024-06-24:
IVe added that.
Can you do me a favor and provide me the snapshot of your fan that supports the TACH pin? Its just to verify that I show the correct scale.

If you use the print fan use : `http://<PRINTER-ip>/printer/objects/query?fan`

For others replace `fan` with the name of the config entry. (e.g. '[controller_fan my_cool_fan]' -> `http://<PRINTER-ip>/printer/objects/query?controller_fan my_cool_fan`)

closed 2024-06-25 by Clon1998

TheMasterOfTech, 2024-06-26:
@Clon1998 , sorry for the delay. To confirm, my hotend fan is an Orion and it matched what was reported in Mainsail so I think you got this correct and it appears you've closed it. **THANK YOU VERY MUCH AND KEEP UP THE OUTSTANDING WORK!** Love being a lifetime supporter of you, it's a wonderful application!

`{"result": {"eventtime": 58461.439510765, "status": {"heater_fan hotend_fan": {"speed": 1.0, "rpm": 8545.485547345954}}}}`

commit 043f362a (2024-06-24): Fix #377 - Added support for tach/rpm