Clon1998/mobileraker #378: V2.7.0 on iOS 17.5.1 Broken

TheMasterOfTech, 2024-06-26:
Information doesn’t load for the printer. Octoeverywhere connects, but the UI seems to have broken how to interpret the information as all I get is a grey screen after the connection is established. I will see if local will make any difference when I get home but I think it’s something in this latest release that broke it.
[image]

I am running iOS 17.5.1 on an iPhone 12 Pro with Mobileraker version 2.7.0.

Clon1998, 2024-06-26:
Hey,
can you provide me the app's logs?
They are located at the very bottom of the app settings.

TheMasterOfTech, 2024-06-26:
@Clon1998 , I don’t see a way to export out the log other than viewing it. Do I need to take snapshots and upload here?

Clon1998, 2024-06-26:
I've already located the issue. It was caused by an error in the remote announcements. It might take a few hours for the new changes to be published on your device. If the error persists after 24 hours, please let me know!

To share the logs, you should be able to press "Share" in the log dialog.

TheMasterOfTech, 2024-06-26:
> I've already located the issue. It was caused by an error in the remote announcements. It might take a few hours for the new value to be published on your device. If the error persists after 24 hours, please let me know!
> 
> To share the logs, you should be able to press "Share" in the log dialog.

@Clon1998 , I don’t see a Share button within the prompt, unless I’m not understanding?

Clon1998, 2024-06-26:
> @Clon1998 , I don’t see a Share button within the prompt, unless I’m not understanding?

https://github.com/Clon1998/mobileraker/assets/10357775/c21943db-c74d-4880-8536-b6d5475c8344

TheMasterOfTech, 2024-06-26:
> > > I've already located the issue. It was caused by an error in the remote announcements. It might take a few hours for the new value to be published on your device. If the error persists after 24 hours, please let me know!
> > 
> > 
> > > 
> > 
> > 
> > > To share the logs, you should be able to press "Share" in the log dialog.
> > 
> > 
> > @Clon1998 , I don’t see a Share button within the prompt, unless I’m not understanding?
> 
>  trim.3E456842-15A1-430E-9246-9155EE2C3B2C.MOV

Didn’t think it was the save button but makes sense lol. Sorry about that! Here you go in case you find something else:
[mobileraker_2024-06-26T10-21-45 2.log](https://github.com/user-attachments/files/15993705/mobileraker_2024-06-26T10-21-45.2.log)
[mobileraker_2024-06-26T11-01-34 2.log](https://github.com/user-attachments/files/15993706/mobileraker_2024-06-26T11-01-34.2.log)
[mobileraker_2024-06-26T10-22-09 2.log](https://github.com/user-attachments/files/15993708/mobileraker_2024-06-26T10-22-09.2.log)
[mobileraker_2024-06-26T10-34-24 2.log](https://github.com/user-attachments/files/15993709/mobileraker_2024-06-26T10-34-24.2.log)
[mobileraker_2024-06-26T10-19-18 2.log](https://github.com/user-attachments/files/15993710/mobileraker_2024-06-26T10-19-18.2.log)
[text 2.txt](https://github.com/user-attachments/files/15993712/text.2.txt)

spitfire, 2024-06-26:
Same on 18b2. On he bright side the live activity started working again;)

TheMasterOfTech, 2024-06-27:
Issue seems resolved, but be it known that I did have to uninstall and reinstall in order to get “the fix” as it might be something cached and was using old configuration.
[image]

closed 2024-06-27 by TheMasterOfTech

Clon1998, 2024-06-27:
> Issue seems resolved, but be it known that I did have to uninstall and reinstall in order to get “the fix” as it might be something cached and was using old configuration.
> 
> [image]
> 
> 

It was related to my remote in app announcements.
They updated at most every 12hrs. So it just took some time. I also just updated the implementation to work in real time to ensure such an issue can be fixed faster next time.