Clon1998/mobileraker #380: App build error

parkkh12, 2024-06-28:
## Feature Request

### Problem Description
I updated Mobileraker_pro to the latest version, completely removed the project, and then recreated it following the procedures. However, during the build process, while the import is recognized normally, the functions inside Mobileraker_pro are not recognized.

### Proposed Solution
What does the following comment in pubspec.yaml mean: "#path: ../mobileraker_pro_public # Uncomment this line if you want to contribute to the project. Also remove the line above"? Is this comment related to the issue?

### Alternatives Considered
If you've thought of any alternative solutions or features, please briefly describe them here.

### Additional Context
I want to contribute by translating to Korean and am working on the Korean translation. I need to build and use the app myself to check for any errors in the translation, but the app does not build at all. What should I do? Below are my configuration screenshots.

Add any screenshots, examples, or additional context to help us understand your feature request better.

[image]

[image]

[image]

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ ] Provided a clear description of the problem you're facing.
- [ ] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2024-06-28:
See the step-by-step [guide](https://github.com/Clon1998/mobileraker?tab=readme-ov-file#environment-setup-for-contribution).

You will need to clone the `mobileraker_pro_pub` repo and use the correct folder structure. Afterwards, update the `mobileraker_pro` dependency to use the folder/repo `mobileraker_pro_pub`.

parkkh12, 2024-06-28:
@Clon1998  I'm really sorry, but could you please explain in detail how to set it up? I'm not a developer and I'm using Flutter for the first time as a hobbyist.

Clon1998, 2024-06-28:
> @Clon1998 I'm really sorry, but could you please explain in detail how to set it up? No matter how closely I follow the guide, I encounter an error with pro_pub. I'm not a developer and I'm using Flutter for the first time as a hobbyist.

Sure!
I'll do a more detailed write up once I am back home. I hope I can do it today but I might need to move it to Sunday.

parkkh12, 2024-06-28:
> Sure! I'll do a more detailed write up once I am back home. I hope I can do it today but I might need to move it to Sunday.

Thank you, I would like to ask for a favor.

Clon1998, 2024-06-29:
Hey,
I hope this helps. Its a bit more detailed compared with the instructions from the main README.

## 1. Clone the `mobileraker` Repository

You will need to clone the `mobileraker` Repository.

### How to Clone:

1. **Install Git:** If you don't have Git installed, download it from Git’s website.

2. **Clone the Repository:**
   - Open your terminal or command prompt.
   - Navigate to the location where you want to store the projects. For example:
     ```bash
     cd path/to/your/projects
     ```
   - Run the following command to clone the `mobileraker` repository:
     ```bash
     git clone https://github.com/Clon1998/mobileraker.git
     ```
     
## 2. Clone the `mobileraker_pro_pub` Repository

You need to clone another project that the mobileraker project depends on.

### How to Clone:

1. **Install Git:** If you don't have Git installed, download it from Git’s website.

2. **Clone the Repository:**
   - Open your terminal or command prompt.
   - Navigate to the location where you want to store the projects. For example:
     ```bash
     cd path/to/your/projects
     ```
   - Run the following command to clone the `mobileraker_pro_pub` repository:
     ```bash
     git clone https://github.com/Clon1998/mobileraker_pro_pub.git
     ```

## 3\.  Check Folder Structure:
   Make sure your folders are organized like this:
   ```plaintext
   .
   ├── mobileraker
   └── mobileraker_pro_pub
```

4\. Configure Dependency in `pubspec.yaml`
------------------------------------------

Now you need to tell Flutter where to find the `mobileraker_pro_pub` project.

### Steps to Configure:

1.  Open the `pubspec.yaml` file located in the `mobileraker` project folder.
2.  Find the section that starts with `dependencies:` and make sure it looks like this:

```yaml
dependencies:
    ...
    mobileraker_pro:
        path: ../mobileraker_pro_pub
```

This tells Flutter that the `mobileraker_pro` dependency is located in the `mobileraker_pro_pub` folder that is one level up from the `mobileraker` folder.

5\. Generate Required Files
---------------------------

You need to generate some files that are necessary for the app to work correctly.

### How to Generate Files:

1.  Open the terminal or command prompt.

2.  Navigate to the root folder of the `mobileraker` project:

```bash
cd path/to/your/projects/mobileraker
```

3.  Run the following commands one by one:

```bash
flutter pub get
flutter packages pub run build_runner build
```
-   `flutter pub get` fetches the dependencies listed in `pubspec.yaml`.
-   `flutter packages pub run build_runner build` generates files needed for the app.

4.  Navigate to the `common` folder:

```bash
cd common
```

5.  Run the following commands:

```bash
flutter pub get
flutter packages pub run build_runner build
```

6\. Configure Firebase
----------------------

Firebase provides backend services for your app, such as authentication and database.

### Steps to Configure Firebase:

1.  Open Your Terminal or Command Prompt.
2.  Run the Command:

```bash
flutterfire configure
```

3.  Follow the instructions in the terminal to configure Firebase for both Android and iOS platforms.

7\. Update `firebase_options.dart`
----------------------------------

After configuring Firebase, you'll have a file named `firebase_options.dart`.

### How to Update:

1.  **Find the `firebase_options.dart` File:** This file is located in the `mobileraker_pro_pub` folder. Open it with your IDE.
2.  **Copy the Content:** Copy the contents of the `firebase_options.dart` file that was generated by the `flutterfire configure` command.
3.  **Replace the Existing File:** Replace the contents of the `firebase_options.dart` file in `mobileraker_pro_pub` with the new contents.
4.  Save the file.

8\. Run the App
---------------

You are now ready to run the app!

### How to Run the App:

1.  Open Your IDE.
2.  Make Sure Your Device or Emulator is Running.
    -   You can use a physical device or an emulator.
    -   If you don't have an emulator set up, you can create one from the Android Studio or Visual Studio Code.
3.  Run the App:
    -   In Visual Studio Code, click the green play button at the top right.
    -   In Android Studio, click the green play button in the toolbar

parkkh12, 2024-06-29:
> Hey, I hope this helps. Its a bit more detailed compared with the instructions from the main README.
> 
> ## 1. Clone the `mobileraker` Repository
> You will need to clone the `mobileraker` Repository.
> 
> ### How to Clone:
> 1. **Install Git:** If you don't have Git installed, download it from Git’s website.
> 2. **Clone the Repository:**
>    
>    * Open your terminal or command prompt.
>    * Navigate to the location where you want to store the projects. For example:
>      ```shell
>      cd path/to/your/projects
>      ```
>    * Run the following command to clone the `mobileraker` repository:
>      ```shell
>      git clone https://github.com/Clon1998/mobileraker.git
>      ```
> 
> ## 2. Clone the `mobileraker_pro_pub` Repository
> You need to clone another project that the mobileraker project depends on.
> 
> ### How to Clone:
> 1. **Install Git:** If you don't have Git installed, download it from Git’s website.
> 2. **Clone the Repository:**
>    
>    * Open your terminal or command prompt.
>    * Navigate to the location where you want to store the projects. For example:
>      ```shell
>      cd path/to/your/projects
>      ```
>    * Run the following command to clone the `mobileraker_pro_pub` repository:
>      ```shell
>      git clone https://github.com/Clon1998/mobileraker_pro_pub.git
>      ```
> 
> ## 3. Check Folder Structure:
> Make sure your folders are organized like this:
> 
> ```
> .
> ├── mobileraker
> └── mobileraker_pro_pub
> ```
> 
> ## 4. Configure Dependency in `pubspec.yaml`
> Now you need to tell Flutter where to find the `mobileraker_pro_pub` project.
> 
> ### Steps to Configure:
> 1. Open the `pubspec.yaml` file located in the `mobileraker` project folder.
> 2. Find the section that starts with `dependencies:` and make sure it looks like this:
> 
> ```yaml
> dependencies:
>     ...
>     mobileraker_pro:
>         path: ../mobileraker_pro_pub
> ```
> 
> This tells Flutter that the `mobileraker_pro` dependency is located in the `mobileraker_pro_pub` folder that is one level up from the `mobileraker` folder.
> 
> ## 5. Generate Required Files
> You need to generate some files that are necessary for the app to work correctly.
> 
> ### How to Generate Files:
> 1. Open the terminal or command prompt.
> 2. Navigate to the root folder of the `mobileraker` project:
> 
> ```shell
> cd path/to/your/projects/mobileraker
> ```
> 
> 3. Run the following commands one by one:
> 
> ```shell
> flutter pub get
> flutter packages pub run build_runner build
> ```
> 
> * `flutter pub get` fetches the dependencies listed in `pubspec.yaml`.
> * `flutter packages pub run build_runner build` generates files needed for the app.
> 
> 4. Navigate to the `common` folder:
> 
> ```shell
> cd common
> ```
> 
> 5. Run the following commands:
> 
> ```shell
> flutter pub get
> flutter packages pub run build_runner build
> ```
> 
> 6. Navigate to the `mobileraker_pro_pub` folder:
> 
> ```shell
> cd path/to/your/projects/mobileraker_pro_pub
> ```
> 
> 7. Run the following commands:
> 
> ```shell
> flutter pub get
> flutter packages pub run build_runner build
> ```
> 
> ## 6. Configure Firebase
> Firebase provides backend services for your app, such as authentication and database.
> 
> ### Steps to Configure Firebase:
> 1. Open Your Terminal or Command Prompt.
> 2. Run the Command:
> 
> ```shell
> flutterfire configure
> ```
> 
> 3. Follow the instructions in the terminal to configure Firebase for both Android and iOS platforms.
> 
> ## 7. Update `firebase_options.dart`
> After configuring Firebase, you'll have a file named `firebase_options.dart`.
> 
> ### How to Update:
> 1. **Find the `firebase_options.dart` File:** This file is located in the `mobileraker_pro_pub` folder. Open it with your IDE.
> 2. **Copy the Content:** Copy the contents of the `firebase_options.dart` file that was generated by the `flutterfire configure` command.
> 3. **Replace the Existing File:** Replace the contents of the `firebase_options.dart` file in `mobileraker_pro_pub` with the new contents.
> 4. Save the file.
> 
> ## 8. Run the App
> You are now ready to run the app!
> 
> ### How to Run the App:
> 1. Open Your IDE.
> 2. Make Sure Your Device or Emulator is Running.
>    
>    * You can use a physical device or an emulator.
>    * If you don't have an emulator set up, you can create one from the Android Studio or Visual Studio Code.
> 3. Run the App:
>    
>    * In Visual Studio Code, click the green play button at the top right.
>    * In Android Studio, click the green play button in the toolbar

I kept trying, but the same error kept occurring.

I managed to resolve it as follows:

It seems that the dependencies for mobileraker need to be installed last, after common and pro_pub.

The build was successful!

Clon1998, 2024-06-29:
That is a good point.
I'll update the readme and adjust the order of execution.
Please note that some UI elements use placeholders as they are actually implemented ok my own private repo.

closed 2024-06-30 by parkkh12