Clon1998/mobileraker #382: App crashing

igl-x2, 2024-07-08:
After the latest update, the app (on Android 12) now just crashes/shutsdown after a random amount of time.
Before the app actually closes, it becomes very choppy and unresponsive.
Logs are attached 
[mobileraker_2024-07-08T19-19-44.log](https://github.com/user-attachments/files/16134061/mobileraker_2024-07-08T19-19-44.log)
[mobileraker_2024-07-08T19-22-47.log](https://github.com/user-attachments/files/16134062/mobileraker_2024-07-08T19-22-47.log)
[mobileraker_2024-07-08T19-27-41.log](https://github.com/user-attachments/files/16134063/mobileraker_2024-07-08T19-27-41.log)
[mobileraker_2024-07-08T19-28-03.log](https://github.com/user-attachments/files/16134064/mobileraker_2024-07-08T19-28-03.log)
[mobileraker_2024-07-08T19-19-24.log](https://github.com/user-attachments/files/16134065/mobileraker_2024-07-08T19-19-24.log)

Clon1998, 2024-07-08:
Hey,
seems like the issue is caused by the WebRTC library I am using. 
I will see what I can do.

hamid-elaosta, 2024-08-02:
> Hey,
> seems like the issue is caused by the WebRTC library I am using. 
> I will see what I can do.
> 
> 

I was about to raise a separate ticket about Webrtc, but it sounds like you're aware of a potential issue already. When I use mjpeg camera it is pretty stable but since switching to webrtc camera it crashes frequently, almost every time I open it, it'll crash after a few seconds. Quite often it won't connect to the printer or won't load the camera, if I force stop the app it'll work for a short while then the app just instantly closes, it'll keep closing after a few seconds every time I open it. 

Happy to provide some logs etc if it helps.

Clon1998, 2024-08-02:
> I was about to raise a separate ticket about Webrtc, but it sounds like you're aware of a potential issue already. When I use mjpeg camera it is pretty stable but since switching to webrtc camera it [...]

I am still looking into it.  For me and many other IOS users it is working fine.
Maybe more data can help me to resolve this issue therefore, can you provide me:
1. Log files
2. Version of the app
3. Device/OS

hamid-elaosta, 2024-08-02:
> I am still looking into it.  For me and many other IOS users it is working fine.
> Maybe more data can help me to resolve this issue therefore, can you provide me:
> 1. Log files
> 2. Version of the app [...]

I've got the log files but will need to sanitise them when I get back to my computer as they contain PII.

Here's a quick excerpt of the errors I see:

```
19:34:08.343 (+0:00:01.030640)
鈿狅笍 Caught exception while sending ICECandidate to server
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.

19:34:08.350 (+0:00:01.037509)
鈿狅笍 Caught exception while sending ICECandidate to server
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.

19:34:08.350 (+0:00:01.037640)
鈿狅笍 Caught exception while sending ICECandidate to server
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.

19:34:08.350 (+0:00:01.037722)
鈿狅笍 Caught exception while sending ICECandidate to server
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.

19:34:08.350 (+0:00:01.037783)
鈿狅笍 Caught exception while sending ICECandidate to server
19:34:08.413 (+0:00:01.100947)
馃挕 Received a new ICE candidate
19:34:08.415 (+0:00:01.102651)
馃挕 Received a new ICE candidate
19:34:08.415 (+0:00:01.103011)
馃挕 Received a new ICE candidate
19:34:08.417 (+0:00:01.104738)
馃挕 Received a new ICE candidate
19:34:08.418 (+0:00:01.105917)
馃挕 Received a new ICE candidate
19:34:08.418 (+0:00:01.106077)
馃挕 Received a new ICE candidate
19:34:08.420 (+0:00:01.107403)
馃挕 Received a new ICE candidate
19:34:08.420 (+0:00:01.107526)
馃挕 Received a new ICE candidate
19:34:08.420 (+0:00:01.107621)
馃挕 Received a new ICE candidate
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.

19:34:08.430 (+0:00:01.117753)
鈿狅笍 Caught exception while sending ICECandidate to server
MobilerakerDioException [DioExceptionType.badResponse]: This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read more about status codes at https://developer.mozilla.org/en-US/docs/Web/HTTP/Status
In order to resolve this exception you typically have either to verify and fix your request code or you have to fix the server code.
```

App: 2.7.1-511
Device/OS: OnePlus 9 Pro, Android 14

I'll update later with full logs.

Clon1998, 2024-08-02:
> I've got the log files but will need to sanitise them when I get back to my computer as they contain PII.
>
> Here's a quick excerpt of the errors I see:
>
> ```
> 19:34:08.343 (+0:00:01.030640)
> 鈿狅笍 Caught [...]

Thanks, Ill dig into that. But it seems like the Flutter Logs arent helpful at all.
I'll test it on my local android device to see what is going on.

hamid-elaosta, 2024-08-10:
Crowsnest version I'm using is: v4.1.9-1-gd75a3aeb

Clon1998, 2024-08-13:
For anyone experiencing this issue, can you please confirm whether your webcam supports H264 by checking the crownnest.log file?

```log
[08/13/24 12:38:27] crowsnest: Supported Formats:
....
[08/13/24 12:38:27] crowsnest:		[2]: 'H264' (H.264, compressed)
[08/13/24 12:38:27] crowsnest:		Size: Discrete 2560x1440
[08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[08/13/24 12:38:27] crowsnest:		Size: Discrete 1920x1080
[08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
[08/13/24 12:38:27] crowsnest:		Size: Discrete 1280x720
[08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
[08/13/24 12:38:27] crowsnest:		Size: Discrete 640x480
[08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
[08/13/24 12:38:27] crowsnest:		Size: Discrete 640x360
[08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
[08/13/24 12:38:27] crowsnest:		Size: Discrete 320x240
[08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)

```

Rastilyn, 2024-08-30:
> For anyone experiencing this issue, can you please confirm whether your webcam supports H264 by checking the crownnest.log file?
> 
> ```
> [08/13/24 12:38:27] crowsnest: Supported Formats:
> ....
> [08/13/24 12:38:27] crowsnest:		[2]: 'H264' (H.264, compressed)
> [08/13/24 12:38:27] crowsnest:		Size: Discrete 2560x1440
> [08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
> [08/13/24 12:38:27] crowsnest:		Size: Discrete 1920x1080
> [08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
> [08/13/24 12:38:27] crowsnest:		Size: Discrete 1280x720
> [08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
> [08/13/24 12:38:27] crowsnest:		Size: Discrete 640x480
> [08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
> [08/13/24 12:38:27] crowsnest:		Size: Discrete 640x360
> [08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
> [08/13/24 12:38:27] crowsnest:		Size: Discrete 320x240
> [08/13/24 12:38:27] crowsnest:		Interval: Discrete 0.017s (60.000 fps)
> ```

I just finally found this thread, been having the issue myself for a while. I am using a raspberry pi cam 3, i don't even show supported formats in my logs. I did using a previous logitech usb camera. This is everything i show from startup. 

```
[08/26/24 09:37:56] crowsnest: Prepare Startup ...
[08/26/24 09:37:56] crowsnest: INFO: Host information:
[08/26/24 09:37:56] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[08/26/24 09:37:56] crowsnest: Host Info: Release: MainsailOS release 1.1.0 (bullseye)
[08/26/24 09:37:56] crowsnest: Host Info: Kernel: Linux 6.1.21-v7+ armv7l
[08/26/24 09:37:56] crowsnest: Host Info: Model:  Raspberry Pi 3 Model B Plus Rev 1.3
[08/26/24 09:37:56] crowsnest: Host Info: Available CPU Cores: 4
[08/26/24 09:37:56] crowsnest: Host Info: Available Memory: 858828 kB
[08/26/24 09:37:56] crowsnest: Host Info: Diskspace (avail. / total): 20G / 30G
[08/26/24 09:37:56] crowsnest: INFO: Checking Dependencies
[08/26/24 09:37:56] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[08/26/24 09:37:56] crowsnest: Dependency: 'find' found in /usr/bin/find.
[08/26/24 09:37:56] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[08/26/24 09:37:56] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[08/26/24 09:37:57] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[08/26/24 09:37:57] crowsnest: Version Control: ustreamer is up to date. (v5.39)
[08/26/24 09:37:59] crowsnest: Version Control: camera-streamer new version available: (0.2.0) (0.2.0(6062a12)).
[08/26/24 09:37:59] crowsnest: INFO: Print Configfile: '/home/akoval/printer_data/config/crowsnest.conf'
[08/26/24 09:37:59] crowsnest:		[crowsnest]
[08/26/24 09:37:59] crowsnest:		log_path: /home/akoval/printer_data/logs/crowsnest.log
[08/26/24 09:37:59] crowsnest:		log_level: verbose
[08/26/24 09:37:59] crowsnest:		delete_log: false
[08/26/24 09:37:59] crowsnest:		no_proxy: false
[08/26/24 09:37:59] crowsnest:		
[08/26/24 09:37:59] crowsnest:		[cam 1]
[08/26/24 09:37:59] crowsnest:		mode: camera-streamer
[08/26/24 09:37:59] crowsnest:		
[08/26/24 09:37:59] crowsnest:		enable_rtsp: false
[08/26/24 09:37:59] crowsnest:		rtsp_port: 8554
[08/26/24 09:37:59] crowsnest:		port: 8080
[08/26/24 09:37:59] crowsnest:		device: /base/soc/i2c0mux/i2c@1/imx708@1a
[08/26/24 09:37:59] crowsnest:		resolution: 4608x2592
[08/26/24 09:37:59] crowsnest:		max_fps: 40
[08/26/24 09:37:59] crowsnest:		v4l2ctl: AfMode=2
[08/26/24 09:37:59] crowsnest: INFO: Detect available Devices
[08/26/24 09:37:59] crowsnest: INFO: Found 1 total available Device(s)
[08/26/24 09:38:00] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/imx708@1a
[08/26/24 09:38:00] crowsnest: 'libcamera' device(s) resolution(s) :
[08/26/24 09:38:00] crowsnest:		0 : imx708 [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
[08/26/24 09:38:00] crowsnest:		    Colorspace: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
[08/26/24 09:38:00] crowsnest:		                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
[08/26/24 09:38:00] crowsnest:		                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
[08/26/24 09:38:00] crowsnest:		
[08/26/24 09:38:00] crowsnest: 'libcamera' device controls :
[08/26/24 09:38:00] crowsnest:		AfRange (int): min=0 max=2
[08/26/24 09:38:00] crowsnest:		ExposureTime (int): min=13 max=112015443
[08/26/24 09:38:00] crowsnest:		ColourGains (float): min=0.000000 max=32.000000
[08/26/24 09:38:00] crowsnest:		AfMode (int): min=0 max=2
[08/26/24 09:38:00] crowsnest:		AnalogueGain (float): min=1.122807 max=16.000000
[08/26/24 09:38:00] crowsnest:		AeEnable (bool): min=false max=true
[08/26/24 09:38:00] crowsnest:		Saturation (float): min=0.000000 max=32.000000
[08/26/24 09:38:00] crowsnest:		AwbEnable (bool): min=false max=true
[08/26/24 09:38:00] crowsnest:		AwbMode (int): min=0 max=7
[08/26/24 09:38:00] crowsnest:		AfTrigger (int): min=0 max=1
[08/26/24 09:38:00] crowsnest:		Sharpness (float): min=0.000000 max=16.000000
[08/26/24 09:38:00] crowsnest:		Contrast (float): min=0.000000 max=32.000000
[08/26/24 09:38:00] crowsnest:		Brightness (float): min=-1.000000 max=1.000000
[08/26/24 09:38:00] crowsnest:		AfSpeed (int): min=0 max=1
[08/26/24 09:38:00] crowsnest:		AeMeteringMode (int): min=0 max=3
[08/26/24 09:38:00] crowsnest:		NoiseReductionMode (int): min=0 max=4
[... cut ...]
[08/26/24 09:38:00] crowsnest:		AfPause (int): min=0 max=2
[08/26/24 09:38:00] crowsnest:		AeConstraintMode (int): min=0 max=3
[08/26/24 09:38:00] crowsnest:		AfMetering (int): min=0 max=1
[08/26/24 09:38:00] crowsnest:		LensPosition (float): min=0.000000 max=32.000000
[08/26/24 09:38:01] crowsnest: 
[08/26/24 09:38:01] crowsnest: Try to start configured Cams / Services...
[08/26/24 09:38:03] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[08/26/24 09:38:05] crowsnest:  ... Done!
```

Rastilyn, 2024-11-14:
@Clon1998 any updates on this? if there is anything else i could provide to help figure this out please let me know.

Clon1998, 2024-11-14:
> @Clon1998 any updates on this? if there is anything else i could provide to help figure this out please let me know. 

Sorry for my late response. I totally forgot about this issue.
I will have a look during the weekend

Rastilyn, 2024-11-14:
No worries. 

I will also add, i can get the camera to view for about 3 seconds after force closing the app. but then it fails and crashes again shortly after.

Rastilyn, 2024-11-14:
Ok, so really interesting thing i just found out. This is only happening when there is an active print going. Otherwise i can watch the camera without issues. homing, running bed mesh, heating etc. as soon as the print starts it crashes. 

Device: Pixel 8 Pro on Android 15, also had the issue on 6 pro android 14. 

attached should be the rest of the info. app gave a few logs i have attached  them all. 
[image]
[mobileraker_2024-11-14T15-04-48.log](https://github.com/user-attachments/files/17757984/mobileraker_2024-11-14T15-04-48.log)
[mobileraker_2024-11-14T15-10-17.log](https://github.com/user-attachments/files/17757985/mobileraker_2024-11-14T15-10-17.log)
[mobileraker_2024-11-14T15-11-31.log](https://github.com/user-attachments/files/17757986/mobileraker_2024-11-14T15-11-31.log)
[mobileraker_2024-11-14T15-16-11.log](https://github.com/user-attachments/files/17757987/mobileraker_2024-11-14T15-16-11.log)
[mobileraker_2024-11-14T15-16-52.log](https://github.com/user-attachments/files/17757988/mobileraker_2024-11-14T15-16-52.log)
[crowsnest.log](https://github.com/user-attachments/files/17757995/crowsnest.log)

Clon1998, 2024-11-17:
> Ok, so really interesting thing i just found out. This is only happening when there is an active print going. Otherwise i can watch the camera without issues. homing, running bed mesh, heating etc. as soon as the print starts it crashes.
> 
> Device: Pixel 8 Pro on Android 15, also had the issue on 6 pro android 14.
> 
> attached should be the rest of the info. app gave a few logs i have attached them all. ![Screenshot_20241114-150758](https://private-user-images.githubusercontent.com/17017097/386373060-77e35f68-82d7-4479-877e-b16816dd37e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE2MjMzNzUsIm5iZiI6MTczMTYyMzA3NSwicGF0aCI6Ii8xNzAxNzA5Ny8zODYzNzMwNjAtNzdlMzVmNjgtODJkNy00NDc5LTg3N2UtYjE2ODE2ZGQzN2UzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE0VDIyMjQzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAwYTc2YWFjZTAzNmU2NDZhZjRhYzY5ZGM1OWEyNTA4ZjU2NGU1YzJkNTJiNDIzMzE4NWEzNGVmYjFjZTRiMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2LmtZuqeD3r1cnBj2lUnXsfEw8QFZXq78N3Q6toRkxs) [mobileraker_2024-11-14T15-04-48.log](https://github.com/user-attachments/files/17757984/mobileraker_2024-11-14T15-04-48.log) [mobileraker_2024-11-14T15-10-17.log](https://github.com/user-attachments/files/17757985/mobileraker_2024-11-14T15-10-17.log) [mobileraker_2024-11-14T15-11-31.log](https://github.com/user-attachments/files/17757986/mobileraker_2024-11-14T15-11-31.log) [mobileraker_2024-11-14T15-16-11.log](https://github.com/user-attachments/files/17757987/mobileraker_2024-11-14T15-16-11.log) [mobileraker_2024-11-14T15-16-52.log](https://github.com/user-attachments/files/17757988/mobileraker_2024-11-14T15-16-52.log) [crowsnest.log](https://github.com/user-attachments/files/17757995/crowsnest.log)

I believe I may have identified the issue regarding the WebRTC camera. It appears that the camera is being opened and closed too frequently while using the app. Currently, the camera seems to stop and start under the following conditions:

- When the app is first launched
- When changing pages
- When switching between active dashboard tabs
- When the app is sent to the background

This frequent stopping and starting could be causing issues at the OS level, as the WebRTC library I am using serves as a wrapper around the platform code needed to enable WebRTC functionality.

Please keep an eye out for the next update, which is expected in the first week of December. I would appreciate any feedback, as I am unable to replicate the issue on my emulator or physical Android device.

Thank you!

Clon1998, 2024-11-25:
@Rastilyn I just pushed a new beta/open test version 577 (2.8.3). Please let me know if its more stable now.

Rastilyn, 2024-12-02:
apologies, holiday took me away from some things here. Is this an android app version? i have been digging through some of the branches but cant seem to find an APK. also in the beta via play store but i am not on 2.8.3, still showing 2.8.2

Clon1998, 2024-12-02:
@Rastilyn no worries!

I published a new build via GitHub in case the beta channel on google play is not working: 

https://github.com/Clon1998/mobileraker/releases/tag/android2.8.3-beta

Just note that this will disable updates so you might want to switch back to google play afterward.

Rastilyn, 2024-12-02:
So far so good! Camera loaded faster than normal as well and have been excessively watching without a crash for the past hour.

The beta did post to the play store as well just fyi, I have to force a update check a few times after joining the beta

Clon1998, 2024-12-14:
Going by the silence, I assume the issue seems to be resolved.
Therefore, I am gone close it but feel free to comment/reopen it if the issue is still present!

closed 2024-12-14 by Clon1998