Clon1998/mobileraker #384: ETA is wrong on app and widget.

BlackNet, 2024-07-10:
The printer shows the previous prints are 7 hours and 1 minute.  I started this current job at 3pm (EST), one hour has elapsed.  The ETA panel on the web interface shows 10:43 PM (close to accurate), while the time on the app shows 2:22 AM.

This print was setup with ocaslicer.  Is there something i can do to get that to read the correct ETA? The mobileraker conf shows EST. Wondering if this is an orca slicer thing.

BlackNet, 2024-07-10:
it is in the app AND on the widget on the lock screen as well.

Clon1998, 2024-07-10:
Alright,
there is not much else you can do.
I need to look into this, but mobileraker uses the same formula as mainsail (Unless they changed it again).
Can you compare all ETA values (Long press the ETA and the app shows you each category).

BlackNet, 2024-07-10:
Hmm. Looks like the slicer and file is correct. The file is what is wrong and it seems to be using that. 

[image]

Clon1998, 2024-07-10:
That looks fine,
Can you tell me what source you enabled for the ETA calculation in your Webinterface? Also are you using Mainsail or Fluidd?
To see the selected sources in Fluidd: Settings -> General -> Print ETA calculation
To see the selected sources in Mainsail: Settings -> General -> Print ETA calculation

BlackNet, 2024-07-10:
I have both mainsail and fluidd on this machine.  I have fluidd set to relative file position, mainsail to file, filament, slicer.

Clon1998, 2024-07-10:
In that case mainsail should also be off as mobileraker currently uses all available sources.

BlackNet, 2024-07-10:
[image]
[image]

This is what each shows.

BlackNet, 2024-07-10:
I guess this swing would be different in several cases. Making it configurable would be the best. That way, each user could change it according to what they are printing.

Clon1998, 2024-07-17:
2.7.1 added that option

closed 2024-07-17 by Clon1998