Clon1998/mobileraker #385: Android notifications - differentiate between progressbar and progress

HoreaM, 2024-07-12:
## Feature Request

### Problem Description
We currently have 3 types of notifications on Android (M117, Progress and Status).
We have the option in the app to add a progressbar notification or a progress notification that updates and a set interval.
The issue is, both the progressbar and the progress notification fall under the same category in Android (Progress).
I would love the ability to make the progressbar notification silent in Android and keep the progress notification as not silent.

### Proposed Solution
Just to simply differentiate between the two notifications (progressbar and progress) and make an additional Android notification category, so that they don't both fall in the same "Print Progress Updates" category in Android would be great!

Clon1998, 2024-07-12:
Hey, that's a good point. In the survey I conducted, improving the notification configuration is the Android feature with the most votes.

Clon1998, 2024-07-30:
Hey,
I just added a separate notification channel for it. However, I am confused in regards to the user ability to configure it.
I think the current settings allow an Android user to configure if he wants to:
a) Use Progres notifications, Text bases
b) Use Progres bar notifications
c) a & b at the same time

Or am I missing something/ is the. Current UI not as easy to understand?

You're feedback on that would be awesome.

HoreaM, 2024-07-31:
Yes, that's correct, we can enable either progress (text) notifications, progress bar notifications or both of them. 
My issue was that both notifications shared the same notification channel. That means, if I had both notification types active (bar & text), I couldn't for example make the progress bar notification be a silent notification in android and leave the progress (text) notification vibrate whenever I receive it.

Did you already release the changes? I still don't have a separate channel for the progress bar or progress text notifications. I'm on the 2.7.1-511 version of the app.

Clon1998, 2024-07-31:
> Yes, that's correct, we can enable either progress (text) notifications, progress bar notifications or both of them.
>
> My issue was that both notifications shared the same notification channel. That [...]

Thanks for the feedback!
The separate notification category was added a few days ago to the Dev version, I will release it probably in the next 1-2 weeks depending on how much stuff I get done.

HoreaM, 2024-07-31:
That sounds great, thanks!

HoreaM, 2024-09-30:
I was able to test this in the last version and it works perfectly, thanks a lot!

closed 2024-09-30 by HoreaM