Clon1998/mobileraker #391: No notifications from printer in Android App

opc241, 2024-07-27:
Hi Clon1998, 

first let me say thank you for you work on this great app! 

Second: What would be the point to check if exactly no notification is sent by Mobileraker (Android OS)? 
I already reinstalled the app ... without success. 
Mobileraker Companion is successfully installed on the printer. I can also remotely use the printer properly. 

Thanks in advance! 

BR OPC

Clon1998, 2024-07-27:
Hey,
here is a list of things you can try:

1. Check if the `mobileraker_companion` service is running
You can do that via ssh and the `ps -ef | grep mobileraker.py`:
```sh
pi@ratty:~ $ ps -ef | grep mobileraker.py
pi         706     1  0 Jul25 ?        00:23:07 /home/pi/mobileraker-env/bin/python3 /home/pi/mobileraker_companion/mobileraker.py -l /home/pi/printer_data/logs -c /home/pi/printer_data/config/mobileraker.conf
pi       11490 11251  0 13:15 pts/0    00:00:00 grep --color=auto mobileraker.py
```

2. Check the `mobileraker.log` file (You can also provide them to me). It should state if it submits noticiations or not:
```log
2024-07-23 12:58:36,616 mobileraker.V2_1111 INFO - 1 Notifications for machineID: 9e4b958c-8dd8-4a87-a256-1cf2c1f3840f: state: False, proggress(text): False, M117 False, GcodeResponse: False, LiveActivity: False, FilamentSensor: 0, progressbar(android): True
2024-07-23 12:58:36,617 mobileraker.V2_1111 INFO - Updating snap in FCM Cfg for 9e4b958c-8dd8-4a87-a256-1cf2c1f3840f: NotificationSnap(progress=30, progress_live_activity=30, progress_progressbar=30, state=printing, m117=f69b1c98bd374e4b8d91e20fc0b918aa4c3f1434a7a5cabd348a4a4a4a0fa6fc, gcode_response=ebb6ec8dde9f7762dde4ee8dd13767b676d1b3a7c96e1b3a5089669f08189052, filament_sensors=[])
2024-07-23 12:58:36,640 mobileraker.V2_1111 INFO - Updated snap in FCM Cfg for 9e4b958c-8dd8-4a87-a256-1cf2c1f3840f: {'jsonrpc': '2.0', 'result': {'namespace': 'mobileraker', 'key': 'fcm.9e4b958c-8dd8-4a87-a256-1cf2c1f3840f.snap', 'value': {'progress': 0.3, 'progress_live_activity': 0.3, 'progress_progressbar': 0.3, 'state': 'printing', 'm117': 'f69b1c98bd374e4b8d91e20fc0b918aa4c3f1434a7a5cabd348a4a4a4a0fa6fc', 'filament_sensors': [], 'gcode_response': 'ebb6ec8dde9f7762dde4ee8dd13767b676d1b3a7c96e1b3a5089669f08189052'}}, 'id': 2308}
2024-07-23 12:58:36,641 mobileraker.cam INFO - Trying to take a snapshot from URI: http://127.0.0.1/webcam/?action=snapshot
2024-07-23 12:58:37,052 mobileraker.cam INFO - Took webcam snapshot! Rotating it using rotation 180°
2024-07-23 12:58:37,059 mobileraker.fcm INFO - Submitting 3 device-requets to mobileraker server
```

3. To trigger notifications without a print job, use [custom notifications](https://github.com/Clon1998/mobileraker_companion/blob/main/docs/Custom_Notifications.md).

opc241, 2024-07-27:
Ok that's strange. I installed the mobileraker companion with the "Install Helper Script" and there is also a green checkmark that confirms the installation of the mobileraker companion, but if i run the command, the service seems not to be running.

`root@K1C-F8E0 / [#] ps -ef | grep mobileraker.py
14691 root      0:00 grep mobileraker.py
`

opc241, 2024-07-27:
Edit: I installed it with your instructions and now it's working. Thank you!

closed 2024-07-27 by opc241