Clon1998/mobileraker #397: Black page with when T0 macro is defined

noobydp, 2024-08-21:
S## Bug Report

### Description
I use the toolchange extension: https://github.com/viesturz/klipper-toolchanger with 2 tools defined and I noticed when I add the [gcode_macro T0] (or T1, T2 etc) section to my config, the extruder section in the app shows a grey screen.

Removing the section and restarting the app allows it to show the controls, with the option to swap between extruders.

This isnt high priority for me as I'm able to remove that section from my config, but I was half way through writing this ticket when I figured out what exactly was causing the bug so I figured I'd submit anyway.

```
[tool T0]
tool_number: 0
extruder: extruder
gcode_x_offset: 0.0
gcode_y_offset: 0.0
gcode_z_offset: 0.0
# # params_servo_angle: 100
params_park_x: 199
params_park_y: -63
# params_park_retract_mm: 5
# params_park_retract_speed: 20
# # params_input_shaper_freq_x:
# # params_input_shaper_freq_y:
# # params_shaper_type_x:
# # params_shaper_type_y:

# [gcode_macro T0]
# variable_color: ""
# gcode:
#   SELECT_TOOL T=0

```

[config_with_tool.txt](https://github.com/user-attachments/files/16688200/config_with_tool.txt)

### Steps to Reproduce
1. add the [gcode_macro T0] to config, firmware restart and kill app
2. go to the second tab
3. scroll to the extruder section

### Expected Behavior
The extruder controls should appear

### Screenshots
Grey box (error)
[image]

Expected controls
[image]

### Version Information

- Device-OS (Android/iOS): iOS
- Mobileraker: 2.7.1-511
- Klipper: v0.12.0-286-g81de9a86
- Moonraker: v0.9.2-6-gf735c041

### Debug Logs
[Logs.zip](https://github.com/user-attachments/files/16688251/Logs.zip)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x ] Provided a clear bug description.
- [ x] Listed detailed steps to reproduce the issue.
- [x ] Described the expected behavior.
- [ x] Included the Mobileraker version you are using.
- [x ] Attached Mobileraker's debug log files.
- [x ] Specified the version numbers of Klipper and Moonraker if applicable.
S

Clon1998, 2024-08-21:
Hey there, thanks for informing me. I recently added the tool selection feature, but I was never able to verify if it works with a real tool changer. Instead, I implemented it based on fluids/mainsail's implementation. I will look into this either later today or next week.

noobydp, 2024-08-21:
That's awesome, thank you.

Clon1998, 2024-08-21:
I have reviewed the provided logs. Unfortunately, due to an error on my part, UI errors are not currently being logged. It appears that you have disabled crashalytics reporting, as I was unable to find anything in my dashboard. Could you kindly do me a favor and either enable temporary crashalytics logging or provide me with the output of the following Moonraker-API endpoint of your printer?

<YOUR-Printer-IP>/printer/objects/query?gcode_macro%20T0

noobydp, 2024-08-22:
I have analytics enabled since I reinstalled the app yesterday. If this still doesn't work I can probably look at compiling the app from source over the weekend and directly debugging... or just send anything else you need.

`{
    "result": {
        "eventtime": 218024.23440438,
        "status": {
            "gcode_macro T0": {
                "color": ""
            }
        }
    }
}`

The actual cfg file has the macro defined like this. I copied it from an example, I dont actually use the variable anywhere.
`[gcode_macro T0]
variable_color: ""
gcode:
  SELECT_TOOL T=0`

Clon1998, 2024-08-22:
Thank you, that was already helpful. The variable is set to an empty string, which leads to an error on my end. I am attempting to retrieve the integer from the hex color, but because it's empty, an error occurs.
I'll commit a fix asap

Clon1998, 2024-08-22:
> I'm not sure what its actually used for.
I believe it is used to facilitate the identification of your tools if the toolhead supports it. I have just implemented the fix that should address that issue. As you have already noticed, for the time being, you can leave it empty or set the color variable to an actual hex value to prevent the app from displaying the gray screen until I deploy the fix.

commit 40126735 (2024-08-22): Fix #397 - Ensured an empty color var wont crash the UI
closed 2024-08-26 by Clon1998