Clon1998/mobileraker #409: Mobileraker is now showing camera, menu icon and other section icons when in printing

salbang, 2024-09-08:
## Bug Report

### Description
After upgrading to the latest version, both in iphone and android, mobileraker does not show camera and icons in dashboard while in printing.

### Steps to Reproduce
1. Overview -> Select printer in printing

2. Then the page is like as attached images.

### Expected Behavior
Dashboard should show camera and icons. 

### Screenshots
If applicable, add screenshots to help explain the problem.
iOS
[image]
[image]

Android
[image]
[image]

### Version Information

- Device-OS (Android/iOS): Both Android and iOS
- Mobileraker: 2.7.2 for both
- Klipper: v0.12.0-290
- Moonraker: v0.9.3

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.
iOS
[mobileraker_2024-09-08T12-38-16.log](https://github.com/user-attachments/files/16920798/mobileraker_2024-09-08T12-38-16.log)
[mobileraker_2024-09-08T12-36-27.log](https://github.com/user-attachments/files/16920799/mobileraker_2024-09-08T12-36-27.log)
[mobileraker_2024-09-08T12-33-57.log](https://github.com/user-attachments/files/16920800/mobileraker_2024-09-08T12-33-57.log)
[mobileraker_2024-09-08T11-56-05.log](https://github.com/user-attachments/files/16920801/mobileraker_2024-09-08T11-56-05.log)

Android
[mobileraker_2024-09-08T11-52-42.log](https://github.com/user-attachments/files/16920837/mobileraker_2024-09-08T11-52-42.log)
[mobileraker_2024-09-08T11-54-39.log](https://github.com/user-attachments/files/16920842/mobileraker_2024-09-08T11-54-39.log)
[mobileraker_2024-09-08T12-39-21.log](https://github.com/user-attachments/files/16920845/mobileraker_2024-09-08T12-39-21.log)
[mobileraker_2024-09-08T12-43-32.log](https://github.com/user-attachments/files/16920846/mobileraker_2024-09-08T12-43-32.log)
[mobileraker_2024-09-08T12-45-15.log](https://github.com/user-attachments/files/16920847/mobileraker_2024-09-08T12-45-15.log)

### Additional Context
Add any other relevant context about the problem here.
The problem did not exist before upgrading to 2.7.2 version.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.

- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-09-08:
Hey,
thanks for reporting this issue. I already saw it popping up in my error reporting and fixed it in 98c4806f1a7189059ab4e45b42fcd94eb5ebd8d0.

2.7.4 should be rolling out since yesterday. It is already available in the open beta on Android.

Clon1998, 2024-09-08:
Can you do me a favor and try 2.7.4 via the testflight version (https://testflight.apple.com/join/h9JwuyR0) and report back to me if the issue is resolved.

salbang, 2024-09-08:
[image]

I have checked the testflight, but the version is still 2.7.2 and I still have the same issue.

Clon1998, 2024-09-08:
> ![image](https://private-user-images.githubusercontent.com/12740762/365435971-126d7a6c-db8a-44ea-9fe1-62a73b913358.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Right its WE so apple will approve the version hopefully by monday!

salbang, 2024-09-09:
My android phone got 2.7.4 update yesterday and works fine now. Waiting for iphone update. Thank you.

salbang, 2024-09-10:
My iphone got 2.7.4 testflight update. Everything looks OK. Thank you for the update.

mikec850, 2024-09-10:
Joining the beta to get 2.7.4 fixed the issue for me on Android too.

Clon1998, 2024-09-10:
Awesome!
Will close the issue as 2.7.4 is rolling out slowly until next monday.

closed 2024-09-10 by Clon1998