Clon1998/mobileraker #411: ETA For Progress Loading Bar Doesn't Respect Time Formatting

freddyyeddy, 2024-09-15:
## Bug Report

### Description
The percentage complete (loading bar type) doesn't respect the set time format (Only Shows 24 Hour Time) set in app settings.

### Steps to Reproduce
Set time to 12 hour format and look at loading bar widget it will show 24 hour time instead.
### Expected Behavior
It should respect the format set.

### Version Information

- Device-OS -Android
- Klipper 0.12
- Moonraker 0.9
- Mobileraker (On Printer) 0.4.0
- Mobileraker App (Latest)

Clon1998, 2024-09-16:
I'll have a look at that.
Seems like I do not sync that setting from the app to the companion.

Clon1998, 2024-09-17:
There currently is no sync for the time format. 
To keep it more flexible the companion already supports a user-defined time format.

By default it is set to 24hrs. You can change it by setting the `eta_format ` key.

For more info see the [Config-Ref](https://github.com/Clon1998/mobileraker_companion?tab=readme-ov-file#companion---config)

Clon1998, 2024-09-17:
I closed it to fast. For US you'd set it to `%%m/%%d/%%Y, %%I:%%M:%%S %%p`

closed 2024-09-17 by Clon1998

freddyyeddy, 2024-09-20:
So I tried setting it to "eta-format: %%a %%r" and it doesn't seem to work
I've restarted the service and rebooted the pi/printer. And still no dice.
Am I missing something? I installed it with kamp originally and did double
check there isn't any other config files and there is just the one. Thanks
for your help.

Clon1998, 2024-09-20:
Can you provide the log files and your config of the companion?

freddyyeddy, 2024-09-20:
For Sure Log Files are attached along with a copy of configs. Due to "supported" file types for githup attachments I had to zip them.
[config and logs.zip](https://github.com/user-attachments/files/17077984/config.and.logs.zip)

`[general]
language: en
# !!! DEPRECATED. The app now syncs the app's language to the companion
# one of the supported languages defined in i18n.py#languages (de,en,...)
# !!! For users from the UK: entering 'uk' will resolve to Ukrainian language, not English. Use 'en' for English!
# Default: en
timezone: AKDT
# The system's timezone e.g. Europe/Berlin for Berlin time or US/Central.
# For more values see https://gist.github.com/heyalexej/8bf688fd67d7199be4a1682b3eec7568
# Default: Tries to use system timezone
# Optional
# 24hour time default
# eta_format: %%d.%%m.%%Y, %%H:%%M:%%S
# 12 hour time us date format
# eta-format: %%a %%I:%%M:%%S %%p
eta-format: %%a %%r
# Format used for eta and adaptive_eta placeholder variables
# For available options see https://strftime.org/
# Note that you will have to escape the % char by using a 2nd one e.g.: %d/%m/%Y -> %%d/%%m/%%Y
# Default: %%d.%%m.%%Y, %%H:%%M:%%S
# Optional
include_snapshot: True
# !! SUPPORTER ONLY - This feature requires beeing a supporter of Mobileraker as of now!
# Include a snapshot of the webcam in any print status/progress update notifications
# Default: True
# Optional

[printer default]
moonraker_uri: ws://127.0.0.1:7125/websocket
# Define the uri to the moonraker instance.
# Default value: ws://127.0.0.1:7125/websocket
# Optional
moonraker_api_key: False
# Moonraker API key if force_logins or trusted clients is active!
# Default value: False
# Optional
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
# !! SUPPORTER ONLY - This feature requires beeing a supporter of Mobileraker as of now!
# The ABSOLUT url to the webcam, the companion should make a screenshot of.
# Example MJPEG SnapShot url: http://192.168.178.135/webcam/?action=snapshot
# Optional
snapshot_rotation: 0
# !! SUPPORTER ONLY - This feature requires beeing a supporter of Mobileraker as of now!
# The rotation of the snapshot in degrees
# Default value: 0
# Valud Values: 0,90,180,270
# Optional
ignore_filament_sensors:
# Comma separated list of filament sensors to ignore, ignored filament sensors do not trigger
# a notification if they are triggered. This is useful if you have a filament sensor that is used
# in a MMU setup like ERCF.
# IMPORTANT, do not include the sensor type. E.g. if your sensor is configured
# like: [filament_switch_sensor printhead_sensor] add `printhead_sensor` to the list.
# Default: empty
# Optional`