Clon1998/mobileraker #415: k1max shutdown upon accessing via app on IOS

bodydropped72, 2024-09-20:
## Bug Report

### Description
anytime i access machine remotely after starting a print i notice it takes a while to populate, then machine errors and shuts down and can no longer access ive attached my klippy.log
[klippy (3).log](https://github.com/user-attachments/files/17078780/klippy.3.log)

### Steps to Reproduce
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See the error.

### Expected Behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): [OS]
- Mobileraker: [version]
- Klipper: [version]
- Moonraker: [version]

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-09-20:
Hey,

I'm sorry to hear about the issue. I took a quick look at the klippy file you provided, and it seems to me like the host is overloaded and missing some events from the connected MCUs. I've only seen this on very low-spec hosts, which is unfortunately the case for many Creality Klipper machines. 

```log
[ERROR] 2024-09-20 23:02:41,107 [root] [klippy:invoke_shutdown:300] Transition to shutdown state: {"code": "", "msg":"MCU 'nozzle_mcu' shutdown: Missed scheduling of next digital out event
This is generally indicative of an intermittent
communication failure between micro-controller and host.", "values": []}
```

bodydropped72, 2024-09-20:
I’ve had it work perfectly fine. Maybe a firmware issue after update? I’ll post my issue in the k1max repository thanks for your quick responseRobby Mendel

Clon1998, 2024-12-08:
Closed for now. Reopen if the issue persists

closed 2024-12-08 by Clon1998