Clon1998/mobileraker #418: Spoolman - Load with QR Code

Bupher, 2024-09-28:
## Feature Request

### Problem Description
Manually navigating to filaments through long list can be annoying, especially when discerning between multiples of the same spool.

### Proposed Solution
Add the ability to 'load' a spool to the active printer by scanning the QR code in the mobileraker app.

### Alternatives Considered
Loading the spool manually though the web interface or the app by searching for it. Using some other form of scanner to load the filament.

### Additional Context
Flow chart:
Open mobileraker
Navigate to a machine
Start the physical filament change
Go to spoolman tab
Press 'Load Via QR Code' or similar
Scan spool
Spool is automatically assigned as active on that printer

Potentially add options for loading to a specific nozzle for multi material printers

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ X] Provided a clear description of the problem you're facing.
- [ X] Clearly outlined the proposed solution or feature.
- [ X] Mentioned any alternative solutions or ideas you've considered.
- [ X] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Bupher, 2024-09-28:
Whoops, just a little stupid. Looks like you've got that covered.

closed 2024-09-28 by Bupher

Clon1998, 2024-09-28:
No worries.