Clon1998/mobileraker #419: Spoolman QR Code links won't scan

Bupher, 2024-10-05:
## Bug Report

### Description
QR code scanner for spoolman spools in mobile raker won't scan proprietary link QR codes

### Steps to Reproduce
1. Open mobile raker
2. Attempt to scan a 'URL QR Code' from spoolam
3. Won't scan

### Expected Behavior
I except the QR code to scan and set as active spool just like normal QR codes

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information
- Device-OS (Android/iOS): Android, S22+, One UI 6.1, Android 14
- Mobileraker: 2.8.0-554
- Klipper: 0.12.0-310
- Moonraker: 0.9.3.0-g71f9e677
- Spoolman: 0.20.0 (4f8d5ab)

### Debug Logs
[debug.log](https://github.com/user-attachments/files/17267509/debug.log)

### Additional Context
Add any other relevant context about the problem here.
[image]
[image]

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- [X] Attached Mobileraker's debug log files.
- [X] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-10-06:
Hey,
 I'll look into this. But this should work. Worst case the user will need to define a pattern to match against.

At the same time I will chekc if I can create a deep link to open the app if the device scans such a QR Code as this is also something I want to add via nfc tags

Bupher, 2024-10-06:
I took a screen recording for some more context.

https://youtu.be/dmiYBIu918Y