Clon1998/mobileraker #423: MobileRaker does not support IPv6

aquaratixc, 2024-10-20:
Title: MobileRaker does not support IPv6

Description:

Currently, the MobileRaker application does not support IPv6 addresses. This limitation prevents users from connecting to their 3D printers running Klipper if they are on an IPv6 network.

Steps to Reproduce:

    Ensure your network is configured to use IPv6.
    Attempt to connect to your 3D printer using MobileRaker.
    Observe that the connection fails or the application does not recognize the IPv6 address.

Expected Behavior:
The application should be able to connect to 3D printers using IPv6 addresses.

Actual Behavior:
The application fails to connect to 3D printers when using IPv6 addresses.

Additional Information:

    MobileRaker Version: 2.6.19-440
    Klipper Version: v0.12.0-317-gb89d55238 
    Operating System: android 14

Screenshots/Logs:
[Attach any relevant screenshots or logs that demonstrate the issue]

Possible Solution:
Implement support for IPv6 addresses in the MobileRaker application to ensure compatibility with modern network configurations.

Clon1998, 2024-10-22:
Hey,
I had a look at the code and this requires some bigger refactoring not just in my app but also in one of the libs I am using.

I will keep an eye open for this.

aquaratixc, 2024-10-22:
That would be good. Because IPv6 support would allow using Yggdrasil addresses. For example, in my case, the printer host is behind NAT and the operator does not provide a static IP (and dynamic will not work), but yggdrasil allows you to break through NAT and use the address without any registrations and unnecessary fuss.

jbehrends, 2024-11-16:
Just ran into the same issue when trying to add the IPv6 address of my moonraker install.  
Error message: "This field requires a valid URL address"

+1 to adding IPv6 support

eazrael, 2026-09-19:
+1
Is IPv6 handling so much different?

closed 2026-10-03 by Clon1998