Clon1998/mobileraker #424: More graceful handling of a camera not being available

jbeardon, 2024-10-25:
## Feature Request

### Problem Description
With some regularity, the cameras on my K1 Max (sic) printers become unavailable until reboot (which may not be for days). Mobileraker displays a bunch of errors when this occurs.
### Proposed Solution
Replace camera related errors with a simple 'camera unavailable' indicator.

### Alternatives Considered
Replacing the crappy printers but hey, I'm stuck with them!

### Additional Context
Add any screenshots, examples, or additional context to help us understand your feature request better.

---
[image]

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ ] Provided a clear description of the problem you're facing.
- [ ] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

willkaxxx, 2025-03-13:
Also I would ask to add feature to hide camera if it is not needed. So not to consume resources from the system.

Example from Fluidd

[image]

jbeardon, 2026-10-19:
> 3\. This has bothered me for a long time as well; any updates on this feature being added?  The last (and only) comment was 1.5 years ago for this now 2 year old request.

To be honest, I found this extremely disappointing - especially after paying for the app.
I've now gone all bambu but it remained an issue until I dumped my final klipper printer.

Clon1998, 2026-11-01:
First of all, thanks for becoming a supporter!

Sorry for the very late reply. I work on the app in my free time, and this one slipped through the cracks.

I’ve just updated the loading and error views for webcams so they take up less space. If you don’t mind, could you let me know whether the new layout works better for you?

[image]

[image]