Clon1998/mobileraker #425: Error connecting remotely to octoeverywhere

shimniok, 2024-10-26:
## Bug Report

### Description
Unable to remotely connect to Octoeverywhere. App is displaying an HttpException 604 on launch.

### Steps to Reproduce
1. Set up app for local network access
2. Install Octoeverywhere on klipper host and set up printer on Octoeverywhere website.
3. On app, configure remote connection. Log into OctoEverywhere
4. View dashboard.

### Expected Behavior
Should see normal dashboard

### Screenshots

[image]

### Version Information

- Device-OS (Android/iOS): [OS]
- Mobileraker: 2.8.0-554
- Klipper: Host: 0.12.0-322-g0855994e5
OS: Raspbian GNU/Linux 11 Firmware: v0.12.0-317-gb89d55238 
- Moonraker: v0.9.3-1-g4e00a076
- Octoeverywhere (Klipper add on) 3.6.1-5-ge02d38f3

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

Complete logfile: 
[](https://drive.google.com/file/d/1izgNc712ZxngCCZy1dvMvKLrQ4w7DagM/view?usp=drivesdk)

```
WebSocketException: Connection to 'https://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com:0/websocket#' was not upgraded to websocket
10:21:24.282 (+0:08:31.595772)
💡 [ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket #609659871] IOWebSocketChannel reported NOT READY!
10:21:24.282 (+0:08:31.596121)
💡 [ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket #609659871] WS-Stream closed abnormally!
10:21:24.282 (+0:08:31.596256)
⚠️ [ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket #609659871] Sending GET to https://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket to determine error reason
10:21:24.449 (+0:08:31.763094)
💡 [ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket #609659871] Got Response to determine error reason: 604
10:21:24.449 (+0:08:31.763410)
⛔ [ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket #609659871] WS-Stream error: HttpException: HttpException - StatusCode 604
10:21:24.449 (+0:08:31.763473)
💡 [ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket #609659871] ClientState.connecting ➝ ClientState.error
10:21:24.450 (+0:08:31.763625)
💡 [Klippy Service ClientType.manual@wss://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com/websocket] Received new JRpcClientState: AsyncData<ClientState>(value: ClientState.connecting) -> AsyncData<ClientState>(value: ClientState.error)
10:21:33.745 (+0:08:41.059056)
💡 Pushing route to /printer/edit
10:21:33.866 (+0:08:41.180481)
💡 Permission Permission.location is PermissionStatus.granted
10:22:20.220 (+0:09:27.534077)
💡 Received from Bottom sheet BottomSheetResult{confirmed: true, data: RemoteInterface{remoteUri: https://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com, httpHeaders: {}, timeout: 30, lastModified: null}}
10:22:20.433 (+0:09:27.747103)
👾 tryingToAddOe: false, _remoteInterfaceProvider has value: RemoteInterface{remoteUri: https://shared-oc9jh0gvj8gv0bp6qmr508p8o08ngkor.octoeverywhere.com, httpHeaders: {}, timeout: 30, lastModified: null}
10:22:20.434 (+0:09:27.747720)
👾 tryingToAddRi: true, _octoEverywhereProvider has value: null
10:22:20.434 (+0:09:27.747838)
👾 tryingToAddObico: false, _obicoTunnelProvider has value: null, obicoEnabled:true
10:22:32.531 (+0:09:39.844937)
💡 Navigating to /tool
10:22:35.449 (+0:09:42.763516)
💡 Received fetchAll
10:22:35.460 (+0:09:42.774320)
💡 Received ordering []
10:22:35.460 (+0:09:42.774521)
💡 Received isSupporter false
10:22:35.461 (+0:09:42.774610)
💡 Max allowed machines for non Supporters is 10
10:22:35.461 (+0:09:42.774683)
💡 Updated allMachinesProvider: Ender 3 (1376dc88-eac9-4100-a52c-cdfb6a18133c)
10:22:41.496 (+0:09:48.810345)
💡 Navigating to /
10:22:41.516 (+0:09:48.830433)
💡 Rebuilding _PrinterAppBar for Ender 3
10:22:43.649 (+0:09:50.963550)
💡 [BottomNavigationBar] Page Changed: 1
10:22:43.651 (+0:09:50.964819)
💡 Page Change received: 1
10:22:43.651 (+0:09:50.965139)
💡 DashboardPageController: (aIdx: 0, l:  2) -> (aIdx: 1, l:  2)
10:22:43.664 (+0:09:50.977740)
💡 Rebuilding _PrinterAppBar for Ender 3
10:22:44.891 (+0:09:52.204939)
💡 [BottomNavigationBar] Page Changed: 0
10:22:44.892 (+0:09:52.205866)
💡 Page Change received: 0
10:22:44.892 (+0:09:52.206293)
💡 DashboardPageController: (aIdx: 1, l:  2) -> (aIdx: 0, l:  2)
10:22:44.905 (+0:09:52.219415)
💡 Rebuilding _PrinterAppBar for Ender 3
10:22:56.852 (+0:10:04.166107)
💡 Navigating to /setting
10:22:57.792 (+0:10:05.106193)
⚠️ [Ender 3 (1376dc88-eac9-4100-a52c-cdfb6a18133c)@http://192.168.1.35] Unable to propagate new notification settings because JRPC was not connected!
MobilerakerException{Machine not connected, parentException: null, parentStack: null}
10:22:57.793 (+0:10:05.107079)
⚠️ Error while trying to fetch CompanionMeta for machine Ender 3 (1376dc88-eac9-4100-a52c-cdfb6a18133c)
10:22:59.766 (+0:10:07.080423)
💡 Show Dialog request for DialogType.logging
```

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ X ] Provided a clear bug description.
- [ X ] Listed detailed steps to reproduce the issue.
- [ X ] Described the expected behavior.
- [ X ] Included the Mobileraker version you are using.
- [ X ] Attached Mobileraker's debug log files.
- [ X ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-10-26:
```
604 - App Connection Revoked/Expired
Permanent - The App Connection can no longer be used.
The App Connection was revoked by the user.
The App Connection has expired.
The owner's removed the printer from their account.
The owner's account was deleted.
```

Clon1998, 2024-10-26:
https://octoeverywhere.stoplight.io/docs/octoeverywhere-api-docs/6dd66625f9222-app-connection-usage#octoeverywhere-custom-error-codes

shimniok, 2024-10-26:
I really appreciate the help! All better now. Was definitely PEBKAC :)

closed 2024-10-26 by Clon1998