Clon1998/mobileraker #426: The Raspberry Pi shutdown and restart functions don't work through the confirmation dialog

RK392, 2024-10-30:
## Bug Report

### Description
As of a recent Mobileraker update, not sure which one, pressing the "Shutdown" or "Restart" button followed by "Confirm" stopped working. As a workaround, I discovered that pressing and holding the buttons to skip the confirmation dialog works.

### Steps to Reproduce
1. Tap on the bottom right menu button.
2. Tap "Shutdown" or "Restart".
3. Press "Confirm".
4. Nothing happens.
5. Press "Back".
6. Press and hold "Shutdown" or "Restart".
7. The Pi shuts down or restarts.

### Expected Behavior
Pressing confirm should execute the requested function.

- Device-OS (Android/iOS): Android
- Mobileraker: 2.8.2
- Klipper: v0.12.0-343
- Moonraker: v0.9.3-1

### Additional Context
I've replicated this bug on two different Android phones on two different printers.

Clon1998, 2024-10-30:
Fixed in ios2.8.2 / android2.8.2

closed 2024-10-30 by Clon1998

Clon1998, 2024-10-30:
Sorry for the short answer, I was on the phone.
The longer answer:

I noticed that issue during the weekend and triggered a new release (2.8.2) today. It should start rolling out slowly tomorrow. The Rollout will be complete next Wednesday.

Thanks for taking the time and reporting the issue. Let me know if I can help you otherwise!

RK392, 2024-10-30:
No worries! As soon as I got your reply, I opened the Play Store and downloaded the update (I'm a beta tester). Now, everything works as it should.

Thank you for actively developing such a useful app!