Clon1998/mobileraker #427: Failed-Key: bed_mesh "default"

kiinami, 2024-11-01:
## Bug Report

### Description
Connecting the printer gives error on loading the dashboard. Console and other sections work correctly.

### Steps to Reproduce
1. Connect the printer
2. Enter the dashboard

### Expected Behavior
Dashboard loads

### Version Information

- Device-OS (Android/iOS): iOS
- Mobileraker: 2.8.1-563

### Debug Logs
Exception:
 NoSuchMethodError: Closure call with mismatched arguments: function 'PrinterBuilder._updateBedMesh'
Receiver: Closure: (Map<String, dynamic>, PrinterBuilder) => PrinterBuilder from Function '_updateBedMesh@855022109': static.
Tried calling: PrinterBuilder._updateBedMesh("\"default\"", _Map len:0, Instance of 'PrinterBuilder')
Found: PrinterBuilder._updateBedMesh(Map<String, dynamic>, PrinterBuilder) => PrinterBuilder

#0      Object.noSuchMethod (dart:core-patch/object_patch.dart:38)
#1      _objectNoSuchMethod (dart:core-patch/object_patch.dart:85)
#2      PrinterBuilder.partialUpdateField (package:common/data/dto/machine/printer_builder.dart:219)
#3      PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:530)
#4      PrinterService._temperatureStore.<anonymous closure> (package:common/service/moonraker/printer_service.dart:484)
#5      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:633)
#6      PrinterService._parseQueriedObjects (package:common/service/moonraker/printer_service.dart:545)
#7      PrinterService._printerObjectsQuery (package:common/service/moonraker/printer_service.dart:569)
<asynchronous suspension>
#8      PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:172)
<asynchronous suspension>

Failed-Key: bed_mesh "default" 
Raw JSON:
```json
{
    "bed_mesh": {
        "mesh_matrix": [
            [
                0.361349,
                0.246143,
                0.165415,
                0.108849,
                0.067822,
                0.035402,
                0.006349,
                -0.022888,
                -0.054165,
                -0.087651,
                -0.12182,
                -0.153456,
                -0.177651
            ],
            [
                0.256462,
                0.235537,
                0.203057,
                0.166339,
                0.130803,
                0.09997,
                0.075462,
                0.057004,
                0.042422,
                0.027643,
                0.006696,
                -0.028288,
[... 856 lines cut ...]
        "perimeters": 0,
        "printer_settings_id": "",
        "progress": 0,
        "support_material_auto": 0,
        "timelapse_type": 0,
        "total_layer": 0,
        "total_time": 0,
        "travel_speed": 0
    }
}
```

### Additional Context
The printer is an Anycubic Kobra 3 running custom firmware (https://github.com/utkabobr/DuckPro-Kobra3 and https://github.com/utkabobr/moonraker/tree/duckpro) to enable mobileraker on the heavily modified klipper implementation it has. 

**The error obviously is not the fault of mobileraker but rather of the modified moonraker implementation, so what I would like to know is that is wrong with the JSON it sends so that I can fix it on the CFW side.**

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.

Clon1998, 2024-11-01:
Thanks for reporting the issue.

First time seeing that. I'll have a look during the weekend.

kiinami, 2024-11-01:
> The error obviously is not the fault of mobileraker but rather of the modified moonraker implementation, so what I would like to know is that is wrong with the JSON it sends so that I can fix it on the CFW side.

Please note this

Clon1998, 2024-11-01:
Hey,
so your json contains `"bed_mesh \"default\"": {},` which is causing the issue.
Default Klipper only allows a single `bed_mesh` to be configured; however, your setup allows multiple, which causes an issue with mobileraker's auto-mapping function as it tries to call the `_updateBedMesh` method with the identifier `default,` which is not supported.