Clon1998/mobileraker #429: Adding Creality K2 makes app unusable

jamincollins, 2024-11-11:
## Bug Report

Adding a Creality K2 Plus to Mobileraker results in an ongoing error that makes the app largely unusable.

The display keeps resetting every second or so.  Each reset jumps back to the top of whichever page you're on and blocks the bottom of the page.  So, the problematic printer can't be removed once added, as the removal option is behind the error message.

### Description
Adding a Creality K2 to Mobileraker results in the following exception:

```
Exception:
 MobilerakerException{Fan "chamber_fan" is not a TemperatureFan. Found _$HeaterFanImpl, parentException: null, parentStack: null}

#0      PrinterBuilder._updateTemperatureFan (package:common/data/dto/machine/printer_builder.dart:417)
#1      PrinterBuilder.partialUpdateField (package:common/data/dto/machine/printer_builder.dart:219)
#2      PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:530)
#3      PrinterService._onStatusUpdateHandler.<anonymous closure> (package:common/service/moonraker/printer_service.dart:521)
#4      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:633)
#5      PrinterService._onStatusUpdateHandler (package:common/service/moonraker/printer_service.dart:520)
#6      JsonRpcClient._onChannelMessage.<anonymous closure> (package:common/network/json_rpc_client.dart:328)
#7      Iterable.forEach (dart:core/iterable.dart:347)
#8      JsonRpcClient._onChannelMessage (package:common/network/json_rpc_client.dart:328)
#9      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#10     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:365)
#11     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:297)
#12     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:784)
#13     _StreamController._add (dart:async/stream_controller.dart:658)
#14     _StreamController.add (dart:async/stream_controller.dart:606)
#15     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#16     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:365)
#17     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:297)
#18     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:784)
#19     _StreamController._add (dart:async/stream_controller.dart:658)
#20     _StreamController.add (dart:async/stream_controller.dart:606)
#21     _StreamSinkWrapper.add (dart:async/stream_controller.dart:877)
#22     _GuaranteeSink.add (package:stream_channel/src/guarantee_channel.dart:125)
#23     new AdapterWebSocketChannel.<anonymous closure>.<anonymous closure> (package:web_socket_channel/adapter_web_socket_channel.dart:85)
#24     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#25     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:365)
#26     _DelayedData.perform (dart:async/stream_impl.dart:541)
#27     _PendingEvents.handleNext (dart:async/stream_impl.dart:646)
#28     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:617)
#29     _microtaskLoop (dart:async/schedule_microtask.dart:40)
#30     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)

Failed-Key: temperature_fan chamber_fan 
Raw Json:
{"gcode_move":{"position":[130.903,204.162,23.1,56473.96026999853],"gcode_position":[130.903,204.162,23.1,423.5927699999229]},"print_stats":{"total_duration":12025.806963026,"print_duration":11762.344391066988,"filament_used":56473.96026999853},"virtual_sdcard":{"progress":0.17450453652168393,"file_position":4007719},"temperature_sensor mcu_temp":{"temperature":44.2},"heater_bed":{"temperature":60.03},"temperature_fan chamber_fan":{"temperature":35.52},"motion_report":{"live_position":[174.17981348157937,257.88844611339124,23.179546656104343,49165.413777294874],"live_velocity":107.70453333333333,"live_extruder_velocity":4.888969359943971},"toolhead":{"print_time":94995.23850699356,"estimated_print_time":94992.9338433,"position":[130.903,204.162,23.171894598600513,49176.24032999867]},"extruder":{"temperature":220.31}}
```

### Steps to Reproduce
1. Add a Creality K2 to Mobileraker

### Expected Behavior
Everything works, or at worst it is possible to remove the problematic printer.

### Screenshots
If applicable, add screenshots to help explain the problem.
[image]
[image]

### Version Information

- Device-OS (Android/iOS): Android 12 & 14
- Mobileraker: 2.8.2 
- Klipper: unclear how to retrieve from the K2 Fluidd UI
- Moonraker: unclear how to retrieve from the K2 Fluidd UI

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.
[mobileraker_2024-11-10T17-20-27.log.gz](https://github.com/user-attachments/files/17694158/mobileraker_2024-11-10T17-20-27.log.gz)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-11-11:
Hey,
thanks for reporting this issue. I will look into this to figure out why temperature fans are causing an issue

jamincollins, 2024-11-11:
I'm happy to provide any additional information or config files if necessary.

Clon1998, 2024-11-11:
Sure!
I did not look into it yet but my guess is that creality modified their moonraker/Klipper code which causes trouble now.

Clon1998, 2024-11-11:
Hey,
can you do me a favor and provide me the data of the following endpoints/http requests:

`GET http://192.168.10.34:4408/server/temperature_store` -> open [URL](http://192.168.10.34:4408/server/temperature_store) in browser

`GET http://192.168.10.34:4408/printer/objects/list` -> open [URL](http://192.168.10.34:4408/printer/objects/list) in browser.

Clon1998, 2024-11-12:
I had a look.
The problem is on my end. I thought all `fans` share the same namespace. That means that there can always be one fan for name `chamber_fan`.
This assumption is false as only fans of the same type can not share the same name but you can have one `heater_fan chamber_fan` and one `temperature_fan chamber_fan` both named `chamber_fan`.

I will need to fix this in my code and ensure I dont break any other thing so this might take a while.

If you want to use the app in the mean time, you might want to rename the config of either fan named `chamber_fan`.

Clon1998, 2024-11-12:
TlDr: It did not take that long to test. Fixed it and will include the fix alongside other changes/fixes in the next version!

Thanks for reporting the issue and letting me know about it! This was a rather tricky bug to catch 👍🏻

jamincollins, 2024-11-12:
Thanks for taking a look and fixing it so quickly!

commit 32e315f2 (2024-11-12): fix: Fix #429 Namespace problem for different fan types

The printer builder, object now include the identifier to ensure objects of same class (E.g. fans) with same name can co-exist
commit 239481e5 (2024-12-08): fix: Fix #429 Namespace problem for different fan types

The printer builder, object now include the identifier to ensure objects of same class (E.g. fans) with same name can co-exist
closed 2024-12-08 by Clon1998