Clon1998/mobileraker #430: Swipe left or right on machine title bar doesn't always refresh or show the new selected machine

wikolii, 2024-11-11:
## Bug Report

### Description
When swiping left or right rather fast on the machine title bar to switch to cycle through machines the screen freezes.
It refreshes when tapping the machine name to show the complete list and (haven't checked) maybe also when changing app focus.

### Steps to Reproduce
0. Have multiple machines added
1. Go to 'any individual machine screen'
2. Swipe left or right on 'machine title bar'
3. Randomly see the error, it will still be showing the previous machine but frozen (no position update, temp change, etc)

### Expected Behavior
Show the next machine on the list

### Screenshots
Sorry, I haven't been able to reproduce the bug on purpose

### Version Information

- Device-OS (Android/iOS): IOS
- Mobileraker: 2.8.2-565 (v0.4.0-85-g18064047)
- Klipper: v0.12.0-290
- Moonraker: v0.9.3-0-g71f9e677

### Debug Logs
[mobileraker_2024-11-08T14-17-11.log](https://github.com/user-attachments/files/17704312/mobileraker_2024-11-08T14-17-11.log)
[mobileraker_2024-11-08T16-19-14.log](https://github.com/user-attachments/files/17704313/mobileraker_2024-11-08T16-19-14.log)
[mobileraker_2024-11-09T01-20-52.log](https://github.com/user-attachments/files/17704314/mobileraker_2024-11-09T01-20-52.log)
[mobileraker_2024-11-11T12-11-10.log](https://github.com/user-attachments/files/17704315/mobileraker_2024-11-11T12-11-10.log)

### Additional Context
I think it's been happening since 1 version prior to the current one

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-11-11:
Hey,
just to confirm.
The app is sometimes showing the dashboard/values of a previously selected machine rather then the dashboard/values of the currently selected one?

wikolii, 2024-11-12:
Yes, but only when swiping to change machines and the values are frozen on screen, showing the last "frame", to put a name on it.
Algo just realized that tapping anywhere on the screen unblocks this behavior.
If this is too much of a hassle to solve or you can't even reproduce the bug, as long as it's just me reporting this i can live with it.

Clon1998, 2024-11-12:
> Yes, but only when swiping to change machines and the values are frozen on screen, showing the last "frame", to put a name on it.
>
> Algo just realized that tapping anywhere on the screen unblocks [...]

Okay. I didn't check the logs yet but so far you're the only one reporting it. I'll see if there is something obvious standing out in the logs.