Clon1998/mobileraker #434: Eta date for long prints

Bad-Hackerman, 2024-11-26:
## Feature Request

### Problem Description
When printing long prints the Eta always shows the time when done but not the date, which can become a bit confusing for prints that cross 3 calendar days. 

### Proposed Solution
Give the date of when the print is done as well as the time

### Alternatives Considered
Add a +1 for each additional day from print start 

Awesome project!!

Clon1998, 2024-11-26:
Hey,
thanks for letting me know. Frankly I never noticed that since my prints rarely exceed 1 day.

Just to confirm you are talking about the eta shown in the app on the dashboard?

Not about eta on:
- Notifications 
- GCode page

Bad-Hackerman, 2024-11-27:
Yes the one in the app, not the one in the notification page
[image]

[image]

Forgot to add the screenshot on the original post...

Clon1998, 2024-11-30:
Let me know if this is clear indicator for next `n` days
[image]

commit 949b61c3 (2024-11-27): feat: Adds day indicator

Fix #434
closed 2024-12-08 by Clon1998