Clon1998/mobileraker #436: Defaulting to remote connection and saying not connected

digital0785, 2024-11-29:
## Bug Report

### Description
Trying to figure out an issue. I'm using mobile raker. But it keeps saying octoeverywhere isn't connected to the Internet. But I can connect to the printer via the slicer with the same info. This only happens sporadically but it's super annoying. It also seems to default to remote connection even though my phone is connected to the same network. Klipperdroid doesn't have an issue brining it up but I much prefer the functionality of mobile raker. I've tried turning off smart WiFi switching as well as re setting up the remote connection but no dice 

### Steps to Reproduce
Open app

### Expected Behavior
To open app and load the printer. 

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): [OS]
- Mobileraker: [version]
- Klipper: [version]
- Moonraker: [version]

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---
[image]
[image]
[image]
[image]

[image]

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

digital0785, 2024-11-29:
Ironically I also still get notifications from the printer even though it won't open it. 
[image]

Clon1998, 2024-11-29:
Hey,

first of all can you provide me the app log files as the screenshot of the log only provided 1% of the log context.

Secondly, are you an active user of octoeverywher? Since the octoeverywher dashboard clearly states that your printer was not connected to their service in the last 3 months. If you're no longer using it I suggest you remove the link to octoeverywher.

The reason for the sporadic usage of the remote connection is that whenever an error occurs while trying to connect via the local address, mobileraker will fallback to the remote connection.