Clon1998/mobileraker #440: Feature Request: Notification camera configuration options

thePunderWoman, 2024-12-02:
## Feature Request

### Problem Description
I get nice notifications based on my configuration, and it works great with one exception. It uses the wrong camera to take a photo of the print. I have 2 cameras set up, one for monitoring and one for VAOC calibration. It seems to always pick the VAOC camera, which is required to be configured as camera 1 for it to work. I'm guessing that camera 1 is what is always picked for the notification. It's pretty useless for me, since that cam has a fixed focus and is pointed in the entirely wrong direction to be useful.

### Proposed Solution
Allow us to configure the notifications such that we can specify the camera used, if an image is used at all, etc.

### Alternatives Considered
I've considered ignoring it entirely, which I can do. It's just not optimal.

### Additional Context
Add any screenshots, examples, or additional context to help us understand your feature request better.

Once the next screenshot comes in, I'll come back and post it so you can see what I mean.
---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

thePunderWoman, 2024-12-02:
Another alternative idea: check if a camera is called "chamber" and default to that one rather than number ordering.

Clon1998, 2024-12-02:
Hey,
this is already possible via the config (snapshot uri):

https://github.com/Clon1998/mobileraker_companion

However, I will see how much effort is required to move more of these settings to the app.

thePunderWoman, 2024-12-02:
Oh! Sure enough! I just went and updated it in my config. Thanks! I'll close this issue.