Clon1998/mobileraker #443: Filament Sensor Notification even when Disabled

3DCoded, 2024-12-10:
## Bug Report

### Description

When I go to Mobileraker's Settings and disable "Show Filament Sensor Alerts", I still get filament sensor notifications during a color swap.

### Steps to Reproduce
1. Go to 'Settings'
2. Disable 'Show Filament Sensor Alerts'
3. Physically trigger the filament sensor
4. See the notification.

### Expected Behavior

I expected that when I disabled "Show Filament Sensor Alerts", I would not get notifications about filament sensors triggering

### Screenshots

[image]

[image]
08a5d14)

### Version Information

- Device-OS (Android/iOS): iOS 18.1.1 on iPad Pro with M2
- Mobileraker: 2.8.2-565
- Klipper: v0.12.0-377
- Moonraker: v0.9.3-1

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

[mobileraker-logs.zip](https://github.com/user-attachments/files/18079586/mobileraker-logs.zip)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- [X] Attached Mobileraker's debug log files.
- [X] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-12-10:
Hey,
you just disables the in-app dialogs.

For push notifications you will need to disable the sensor on the companion config.

https://github.com/Clon1998/mobileraker_companion

I know this is not optimal and it's on my to-do list to make these configuration options available directly in the app!

3DCoded, 2024-12-10:
Thank you for the quick reply! I changed the setting and now it works.

closed 2024-12-10 by 3DCoded